jamesremuscat/pyze #1: Bypass environment variables

epenet, 2019-07-29:
Hi,

Do you think it would be possible to pass the Url/ApiKey as parameters instead of Environment variables?

This is what I've done on my [Home Assistant plugin](https://github.com/epenet/hassRenaultZE) for the Url/ApiKey.
`
    async def initialise_configuration(self, language):
        url = 'https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_%s.json' % language
        async with aiohttp.ClientSession(
                ) as session:
            async with session.get(url) as response:
                responsetext = await response.text()
                if responsetext == '':
                    responsetext = '{}'
                jsonresponse = json.loads(responsetext)

                self.initialise_configuration_url(
                    jsonresponse['servers']['gigyaProd']['target'],
                    jsonresponse['servers']['gigyaProd']['apikey'],
                    jsonresponse['servers']['wiredProd']['target'],
                    jsonresponse['servers']['wiredProd']['apikey']
                    )
`

jamesremuscat, 2019-07-29:
Been thinking a bit about this - the original reason for using environment variables was to produce something that people could quickly get working without me reproducing those keys publicly - that's been somewhat obsoleted by Renault doing that themselves!

How would you feel about `set_api_key` methods on `Gigya` and `Kamereon`, and those keys being stored in the PyZE credentials file thereafter? (So you'd only have to call `set_api_key` once, or probably once per login.)

epenet, 2019-07-30:
`set_api_key` sounds good, together with a `set_url` method (or include the url as a second parameter)

I think it would be good to add a method also to load the two url and the two apikey from the AWS server. I'm not sure where that would be sit (Gigya, Kamereon, Credentials or a new class).

Finally, I had issues with the links between Gigya, Kamereon and the Credentials, which doesn't match the README file. It seems to me that Kamereon always creates a Gigya instance, and it's not possible to create one then pass it to Kamereon. In the end, the only way it worked for me with the api was to use the following code:
```
    k = Kamereon()
    k._gigya.login(my_renault_username, my_renault_password)
    k._gigya.account_info()
    v = Vehicle(my_renault_vin, k)
    print(v.battery_status())
```

jamesremuscat, 2019-07-31:
@epenet Would you like to take a look at the API as it stands on `develop`, and let me know if there's anything else you'd need to close this issue?

jamesremuscat, 2019-08-07:
Having merged #7, I'll close this down and do a 0.2.0 release shortly.

closed 2019-08-07 by jamesremuscat