jamesremuscat/pyze #5: Rename project/repository

epenet, 2019-07-30:
Hi @jamesremuscat,
Since we believe this project will be available across the whole Renault range (including non-electric cars), would it make sense to remove references to ZE and replace with MyRenault for example?

jamesremuscat, 2019-07-30:
While the Kamereon API certainly has relevance to non-ZE vehicles, the PyZE API is focused on the ZE parts of it, and I'm not planning to expand it further.

Including a company's registered trademark in this project's name is probably a bad idea :slightly_smiling_face:

closed 2019-07-30 by jamesremuscat

benthehorn, 2019-08-08:
Error communicating with Renault API!
{"type":"FUNCTIONAL","messages":[{"code":"err.func.user.invalid","message":"Kamereon authentication failed! No token ID found."}],"errors":[{"errorCode":"err.func.user.invalid","errorMessage":"Kamereon authentication failed! No token ID found."}],"error_reference":"FUNCTIONAL"}
Is this because I don't yet have a car registered to my account ?

jamesremuscat, 2019-08-08:
@benthehorn Please don't comment on random closed issues if you have a new problem; open a new one instead.