jamesremuscat/pyze #13: WARNING: Multiple Kamereon accounts found. Using the first.

TheReiner, 2019-09-29:
i have this error, i don't know why ? Any Idea ?
Thanks.

root@HomeKit:~# pyze login
Enter your My Renault email address: email@exemple.com                         
Enter your password: 
Logged in successfully.

root@HomeKit:~# pyze status
WARNING: Multiple Kamereon accounts found. Using the first.
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.2.2', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.2.2-py3.7.egg/pyze/cli/__main__.py", line 61, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.2.2-py3.7.egg/pyze/cli/status.py", line 19, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.2.2-py3.7.egg/pyze/cli/common.py", line 41, in get_vehicle
IndexError: list index out of range

epenet, 2019-09-30:
Hi,

It looks like the CLI has trouble finding your vehicle on [line 41 of common.py](https://github.com/jamesremuscat/pyze/blob/develop/src/pyze/cli/common.py)
Is the vehicle/vin correctly registered on your account?

TheReiner, 2019-09-30:
Yes of course... I check ^^
But, if i use your last Script MyRenault.py, it's work fine... 

python3 MyRenault.py 
https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_fr_FR.json
android_config
https://accounts.eu1.gigya.com/accounts.login?loginID=thomas-reiner.roth%40appitel.fr&password=semfuz-Cerwux-jowve2&apiKey=3_4LKbCcMMcvjDm3X89LU4z4mNKYKdl_W0oD9w-Jvih21WqgJKtFZAnb9YdUgWT9_a
gigya_session
https://accounts.eu1.gigya.com/accounts.getAccountInfo?oauth_token=st2.s.AcbHzYRBlQ.Rb7keiobOmGYWzv7w_AfVUmINtWqUx_1vM0lr7_jCG978zKvR64w3rpR6oehm5LszaCDtkX5dhDNeuKHqxSBgfBJcI87pZvnHkwHQynuUug.m-68pdpREiZui5MzUvnLFhSdOuHU_EH7QZFaOPjruaHUzO0I_llFXoO1Vcqij2byYMwCqORQU_dOzKljADdJmg.sc3
gigya_account
https://accounts.eu1.gigya.com/accounts.getJWT?oauth_token=st2.s.AcbHzYRBlQ.Rb7keiobOmGYWzv7w_AfVUmINtWqUx_1vM0lr7_jCG978zKvR64w3rpR6oehm5LszaCDtkX5dhDNeuKHqxSBgfBJcI87pZvnHkwHQynuUug.m-68pdpREiZui5MzUvnLFhSdOuHU_EH7QZFaOPjruaHUzO0I_llFXoO1Vcqij2byYMwCqORQU_dOzKljADdJmg.sc3&fields=data.personId%2Cdata.gigyaDataCenter&expiration=900
gigya_jwt
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/persons/4d7f7d76-e036-4b6a-acbe-36afd8a8d643?country=FR
kamereon_person
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/0447f2a1-276c-408e-96d2-2035a07a8193/kamereon/token?country=FR
kamereon_token
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/XXXX/battery-status
{'data': {'type': 'Car', 'id': 'XXXXX', 'attributes': {'plugStatus': 0, 'chargeStatus': -1, 'batteryLevel': 19, 'lastUpdateTime': '2019-09-30T01:04:45+02:00', 'batteryTemperature': 16, 'rangeHvacOff': 60}}}
root@HomeKit:~# 

And with your new script, i can't set location fr-fr... maybe it's the problem ?

Thanks.

TheReiner, 2019-09-30:
Yes of course... I check ^^
But, if i use your last Script MyRenault.py, it's work fine...

python3 MyRenault.py
https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_fr_FR.json
android_config
https://accounts.eu1.gigya.com/accounts.login?loginID=exemple@exemple.com&password=xxxxxxx&apiKey=xxxx
gigya_session
https://accounts.eu1.gigya.com/accounts.getAccountInfo?oauth_token=xxxx
gigya_account
https://accounts.eu1.gigya.com/accounts.getJWT?oauth_token=xxx&fields=data.personId%2Cdata.gigyaDataCenter&expiration=900
gigya_jwt
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/persons/xxx?country=FR
kamereon_person
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxx/kamereon/token?country=FR
kamereon_token
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/XXXX/battery-status
{'data': {'type': 'Car', 'id': 'XXXXX', 'attributes': {'plugStatus': 0, 'chargeStatus': -1, 'batteryLevel': 19, 'lastUpdateTime': '2019-09-30T01:04:45+02:00', 'batteryTemperature': 16, 'rangeHvacOff': 60}}}
root@HomeKit:~#

And with your new script, i can't set location fr-fr... maybe it's the problem ?

Thanks.

epenet, 2019-09-30:
@TheReiner I suggest you change your Gygia pwd as you may have inadvertently exposed it.

Aside from the above, this warning can be ignored:
  `WARNING: Multiple Kamereon accounts found. Using the first.`
@jamesremuscat maybe we could simply drop the warning? I'm not sure it adds anything.

The real issue is this one, so you should probably update your issue title to something like "CLI issue with pyze status: index out of range"
  `IndexError: list index out of range`

Finally, I don't normally use the CLI, and I am more familiar with the API. I'll see if I find anything, but maybe @jamesremuscat can shed some light...

epenet, 2019-09-30:
@TheReiner, I have tried it all with my account and it's working fine.

Could you try to get the vehicle list from the CLI?
```
  $ pyze vehicles
  Found 1 vehicle
  - XX000XX: RENAULT ZOE [VINVINVINVINVINVIN]
```

Then, can you try to get the status by specifying the VIN?
```
  $ pyze status --vin VINVINVINVINVINVIN
  --------------------  -------------------
  Battery level         52%
  Range estimate        89.5 miles
  Plugged in            No
  Charging              No
  Charge mode           Always charge
  AC state              off
  External temperature  10.0°C
  Total mileage         21075.7 mi
  Updated at            2019-09-30 09:05:51
  --------------------  -------------------
```

jamesremuscat, 2019-09-30:
> Aside from the above, this warning can be ignored:
> `WARNING: Multiple Kamereon accounts found. Using the first.`
> @jamesremuscat maybe we could simply drop the warning? I'm not sure it adds anything.
>
> The real issue is...

@epenet On the contrary, I think this warning might perfectly explain the issue here.

If OP's vehicle is linked to the second Kamereon account, and no vehicles are linked to the first, then this is exactly the behaviour I'd expect to see. I don't know the slightest thing about Renault's CRM data model or why this might be the case. How does your script handle multiple Kamereon accounts on one Gigya account?

@TheReiner I suspect you've encountered a condition that my app doesn't yet handle, because when I was developing it I didn't know if it would actually happen or not! I think I have a quick-and-dirty workaround that I might be able to push later today.

epenet, 2019-09-30:
@jamesremuscat in my original investigations, I found that I had two Kamereon accounts.
With either of them I was able to get my vehicle information.

Now that I use your API, I just have lots of warnings in my logs, which I simply ignore as I didn't find a way to deactivate them.

jamesremuscat, 2019-09-30:
@epenet Ignoring warnings is fine if you're aware of the potential effects - that's why it's a warning not an error.

I've just released v0.3.0 to PyPI and `master` branch that gives you some ways to specify an account ID to use; either via `Kamereon#set_account_id` or a `KAMEREON_ACCOUNT_ID` variable.

CLI users can run `pyze set-account <account-id>` to select a different account if multiple accounts are found.

@TheReiner The new release should give you some more information about what's wrong here. Please could you try running `pyze login` on the new version? It should give you a warning about multiple accounts, and list their IDs.

If `pyze vehicles` then says "Found 0 vehicles", try running `pyze set-account` with the _second_ ID from the list and try again?

jamesremuscat, 2019-12-16:
OP has gone away.

closed 2019-12-16 by jamesremuscat