jamesremuscat/pyze #14: Failed to log in! Error - No Extra Details

edddeduck, 2019-10-11:
I setup a French MyRenault account here:

https://myr.renault.fr

I added my VIN number to my account on the site. I then grabbed the GIGYA + KAMERON ApiKeys from the Renault website and set those up. I then the pyre login:

pyze login
Enter your My Renault email address: [email]
Enter your password: [password]
Failed to log in!

Given the login works fine on the website I am unsure what I have done wrong.

NOTE: I have also installed the French iPhone app and although my car is VIN registered I need to confirm a French phone number before being allowed to activate and use the app to get data. I don't have a French mobile number so I can't complete that part if it's required.

Sorry if I am missing something obvious but I've looked over the code and the blog post and can't see what's going wrong or anyone else with a similar issue that I can use to debug my issue. I assume it's in the account setup which wasn't documented so I am unsure if I have done it wrong or not.

jamesremuscat, 2019-10-12:
It's my assumption that the phone number is a required part of registration before you can use the API.

There are some [online tools](https://www.receive-sms-online.info) that you can use if you don't have a conveniently French phone number already :wink: 

Nevertheless, I should probably give more information in the error message there...

edddeduck, 2019-10-12:
Thanks, I have solved the phone number issue with your link and now have a working French app on my iPhone but the login failure persists.

I did a clean install on Ubuntu. I deliberately used incorrect user/pass at first and didn't define either keys to try and work through errors and see if the original error was something not being defined correctly.

I get an error if I don't define GIGYA but that aside the only error I get is the "Failed to log in!" message. I know the account works on the mobile and on the website. 

**Testing with no defines setup**

pyze$ pyze login
Enter your My Renault email address: test
Enter your password: 
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.3.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.6/dist-packages/PyZE-0.3.0-py3.6.egg/pyze/cli/__main__.py", line 62, in main
  File "/usr/local/lib/python3.6/dist-packages/PyZE-0.3.0-py3.6.egg/pyze/cli/login.py", line 14, in run
  File "/usr/local/lib/python3.6/dist-packages/PyZE-0.3.0-py3.6.egg/pyze/api/gigya.py", line 30, in login
RuntimeError: Gigya API key not specified. Call set_api_key or set GIGYA_API_KEY environment variable.

**Then defined GIGYA**

export GIGYA_API_KEY=[valid key here from website]

**then got** 

pyze$ pyze login
Enter your My Renault email address: test
Enter your password: 
Failed to log in!

**KAMERON hadn't been defined. Defining it didn't make a difference and the error stayed the same.**

export KAMEREON_API_KEY=[valid key here from website]
pyze$ pyze login
Enter your My Renault email address: test
Enter your password: 
Failed to log in!

Happy to try and debug further but my python knowledge is minimal.

edddeduck, 2019-10-12:
I have noticed there is no error messages if the API keys are incorrect.

systems@systems:/etc/openhab2/scripts/pyze$ export GIGYA_API_KEY=CHICKENANDCHIPS
pyze$ pyze login
Enter your My Renault email address: test
Enter your password: 
Failed to log in!

I am using the gigya apikey from the servers section of the Renault configuration/android/config_en_GB.json file on their servers as suggested. I am not using gigyaValid or gigyaProd

epenet, 2019-10-13:
@edddeduck can you try to use the french API keys (from config_fr_FR.json)?
Surely if you are using the French application, and needed a french telephone number, then you will need the french API keys...

Also, please make sure that you are using the gigyaProd and wiredProd api keys :
- gigya_key = jsonresponse['servers']['gigyaProd']['apikey']
- kamereon_key = jsonresponse['servers']['wiredProd']['apikey']

edddeduck, 2019-10-13:
Thanks @epenet that worked, I should have checked the link in the instructions. I can now login but I now see the following error running the status command.

> pyze status
Error communicating with Renault API!
{"type":"FUNCTIONAL","messages":[{"code":"err.func.user.invalid","message":"Kamereon authentication failed! No token ID found."}],"errors":[{"errorCode":"err.func.user.invalid","errorMessage":"Kamereon authentication failed! No token ID found."}],"error_reference":"FUNCTIONAL"}

I'm off to read the python files and see if I can work out what's wrong but any clues would be welcome!

Also I am making notes so I submit a pull request with a few notes to aid installation and highlight the issues I've bumped into being more of a novice with python setup.

jamesremuscat, 2019-10-14:
That's a response from the Renault API itself. For some reason it looks like we weren't able to find a token for your account ([kamereon.py:104](https://github.com/jamesremuscat/pyze/blob/develop/src/pyze/api/kamereon.py#L104)).

If you're running from source you could try adding some `print()` calls to that method and see what the API is actually returning when you try `pyze status`.

> Also I am making notes so I submit a pull request with a few notes to aid installation and highlight the issues I've bumped into being more of a novice with python setup.

Brilliant - thank you!

jamesremuscat, 2019-10-14:
> @edddeduck can you try to use the french API keys (from config_fr_FR.json)?
> Surely if you are using the French application, and needed a french telephone number, then you will need the french API keys...

Interesting point you raise - though it's not the cause of the OP's issues (but rather needing to use the `*Prod` keys). I did a little digging...

Looks like the Kamereon API key is constant across all configuration, but that there are a number of Gigya keys in use. I've tried a handful and they all seem to work interchangeably.

Conversely, none of the Gigya API calls require a `country` parameter, while all of the Kamereon calls require one; I've not been able to discern it having any effect, and indeed `country=XX` works just as well as `country=GB`. :man_shrugging:

jamesremuscat, 2019-12-02:
@edddeduck Are you still experiencing issues?

jrathert, 2019-12-22:
Hi - I had a similar error: The request to get an kamereon token failed, with the response being status 400 and the message saying: `Kamereon authentication failed! No token ID found`. To me, it looks as if the gigya jwt token is not valid for the specific call.

Then I noticed: I did not use your code in the first place, but copied relevant information from **your blog post** into my code. And **there is a typo in the blog post** when fetching the gigya jwt token, where it says: _[...]you need to include data.personId,data.gigyaDaterCenter as the value of fields[...]_. Of course, this should be "data.gigya.**DataCenter**" (not "**DaterCenter**"). After fixing it, my code worked.

Not sure if the original poster did run into a similar problem, probably not, as he was using your code, I guess. But it might give a hint as well: I think the gigya jwt is not appropriate in this case.

jamesremuscat, 2019-12-23:
@jrathert Thanks - I've fixed the typo in the original blog post.

jamesremuscat, 2020-04-24:
OP hasn't replied in more than six months, so I'll close this issue now.

closed 2020-04-24 by jamesremuscat