jamesremuscat/pyze #17: cli: charge-history: KeyError: 'chargeEndDate' / KeyError: 'chargeStartInstantaneousPower'

ArnoutVos, 2019-10-19:
charge-history api call seems to be changed.
When using the cli: charge-history during a actual charge, it gives an error.

```
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.3.0-py3.7.egg/pyze/cli/charge-history.py", line 58, in _format_charge_history
KeyError: 'chargeEndDate'
```
I think the chargeEndDate is empty (because it still runs) and the code expects a date, and doesn't accept nulls?

During non-charging I get this error:

```
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.3.0-py3.7.egg/pyze/cli/charge-history.py", line 67, in _format_charge_history
KeyError: 'chargeStartInstantaneousPower'
```

I tried to fix it... How can I easily display the underlying API response? Or how can I debug a single Python file in this project? I'm running Ubuntu and prefer the CLI as dev env.

jamesremuscat, 2019-10-20:
Thanks for the bug report. The `charge-history` API endpoint has never worked for me (it always returns an empty list) so I'm reliant on reports like this!

If you're able to check out and run the latest code from the `develop` branch, you can pass a new `--debug` flag to the `pyze` command e.g. `pyze --debug charge-history` which will be rather verbose about HTTP calls and response bodies.

I think you're spot on about the cause - the code in https://github.com/jamesremuscat/pyze/blob/develop/src/pyze/cli/charge-history.py#L49 assumes all the dictionary keys are present. It should probably rather use `.get()` to handle the null cases.

ArnoutVos, 2019-10-21:
Thanks for the help.

It turns out my last charge session is not giving the `chargeStartInstantaneousPower`. It seems this is a optional field in the response.

Also, the repsonse indicated it was a 'slow' charge session. I shared the charger with a Tesla and the charger was load-balancing between the Tesla and my Zoe R240.
`pyze status` indicated my car was getting a lousy 6.5 kW (so the charger divided it's power between the Tesla and my Zoe. 

Will try to fix the code and make a pull request if I succeed in that.

```
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1xxx/charges?start=20191001&end=20191021 HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {'data': {'type': 'Car', 'id': 'VF1xxx', 'attributes': {'charges': [{'chargeStartDate': '2019-10-19T12:14:29Z', 'chargeEndDate': '2019-10-19T15:44:20Z', 'chargeDuration': 209, 'chargeStartBatteryLevel': 22, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 78, 'chargePower': 'slow', 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2019-10-19T08:41:49Z', 'chargeEndDate': '2019-10-19T09:59:52Z', 'chargeDuration': 78, 'chargeStartBatteryLevel': 14, 'chargeEndBatteryLevel': 99, 'chargeBatteryLevelRecovered': 85, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 19800, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2019-10-19T08:37:55Z', 'chargeEndDate': '2019-10-19T08:38:52Z', 'chargeDuration': 0, 'chargeStartBatteryLevel': 12, 'chargeEndBatteryLevel': 13, 'chargeBatteryLevelRecovered': 1, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 19700, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2019-10-18T06:17:20Z', 'chargeEndDate': '2019-10-18T07:29:29Z', 'chargeDuration': 72, 'chargeStartBatteryLevel': 59, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 41, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 19700, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2019-10-17T06:26:10Z', 'chargeEndDate': '2019-10-17T08:02:55Z', 'chargeDuration': 96, 'chargeStartBatteryLevel': 21, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 79, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 20200, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2019-10-15T06:34:14Z', 'chargeEndDate': '2019-10-15T08:12:18Z', 'chargeDuration': 98, 'chargeStartBatteryLevel': 15, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 85, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 21800, 'chargeEndStatus': 'ok'}]}}}
```

ArnoutVos, 2019-10-21:
And fixed. Pull request on it's way.

```
pi@raspberrypi:~/Downloads/pyze-develop $ pyze charge-history
Charge start         Charge end           Duration    Power (kW)      Charge gained (%)  Status
-------------------  -------------------  ----------  ------------  -------------------  --------
2019-10-19 14:14:29  2019-10-19 17:44:20  3:29:00                                    78  ok
2019-10-19 10:41:49  2019-10-19 11:59:52  1:18:00     19.80                          85  ok
2019-10-19 10:37:55  2019-10-19 10:38:52  0:00:00     19.70                           1  ok
2019-10-18 08:17:20  2019-10-18 09:29:29  1:12:00     19.70                          41  ok
2019-10-17 08:26:10  2019-10-17 10:02:55  1:36:00     20.20                          79  ok
2019-10-15 08:34:14  2019-10-15 10:12:18  1:38:00     21.80                          85  ok
```

jamesremuscat, 2019-10-23:
Fixed by #18.

closed 2019-10-23 by jamesremuscat