jamesremuscat/pyze #19: KeyError: 'hvacStatus'

mcluki, 2019-12-13:
In a hurry, just dumping this here ;-)

$ pyze vehicles
/usr/lib/python3.8/site-packages/tzlocal-2.0.0-py3.8.egg/tzlocal/unix.py:141: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while start is not 0:
Found 1 vehicle
 - DP40xxx: RENAULT ZOE [VF1AGxxxxxxxxxx]

$ pyze status
/usr/lib/python3.8/site-packages/tzlocal-2.0.0-py3.8.egg/tzlocal/unix.py:141: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while start is not 0:
Traceback (most recent call last):
  File "/usr/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.3.0', 'console_scripts', 'pyze')()
  File "/usr/lib/python3.8/site-packages/PyZE-0.3.0-py3.8.egg/pyze/cli/__main__.py", line 68, in main
  File "/usr/lib/python3.8/site-packages/PyZE-0.3.0-py3.8.egg/pyze/cli/status.py", line 64, in run
KeyError: 'hvacStatus'

jamesremuscat, 2019-12-16:
Thanks for the report.

While it's a simple fix, if you could help me understand what the Renault API is returning here it might be useful!

- Is this reproducible every time?
- If you're running latest `develop` please can you try running `pyze --debug status`, look for the line with `GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/<your VIN>/hvac-status`, and post the response shown on the line below? (Remember to edit out your VIN before posting it)
- What model Zoe do you have?

mcluki, 2019-12-16:
thanks, here u are:

It's a 2015 Q210 model, re-imported from France to "de_AT". But you'll see that - and exciting information about my annualMilage ;-) - in the debugs anyways ;-)

$ pyze --debug status

/usr/lib/python3.8/site-packages/tzlocal-2.0.0-py3.8.egg/tzlocal/unix.py:141: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while start is not 0:
WARNING:pyze:Debug output enabled. Logs may contain personally identifiable information and account credentials! Be sure to sanitise these logs before sending them to a third party or posting them online.
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api-wired-prod-1-euw1.wrd-aws.com:443
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/7b96b187-f958-4d63-93b9-dec3560468c7/vehicles?country=GB HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicles response: {'accountId': '7b96b187-f958-4d63-93b9-dec3560468c7', 'country': 'GB', 'vehicleLinks': [{'brand': 'RENAULT', 'vin': 'VF1AGVYA051xxxxxx', 'status': 'ACTIVE', 'linkType': 'USER', 'garageBrand': 'renault', 'annualMileage': 18000, 'mileage': 57557, 'mileageDate': '2019-12-06', 'estimatedMileage': 57060, 'startDate': '2019-12-06', 'createdDate': '2019-12-06T09:05:24.099252Z', 'lastModifiedDate': '2019-12-06T09:20:23.400311Z', 'cancellationReason': {}, 'preferredDealer': {'dealerId': '12345678_001', 'brand': 'RENAULT', 'createdDate': '2019-12-06T09:08:20.719243Z', 'lastModifiedDate': '2019-12-06T09:08:20.719243Z'}, 'connectedDriver': {'role': 'MAIN_DRIVER', 'createdDate': '2019-12-06T09:10:38.105052Z', 'lastModifiedDate': '2019-12-06T09:10:38.105052Z'}, 'vehicleDetails': {'vin': 'VF1AGVYA051xxxxxx', 'registrationDate': '2015-02-20', 'firstRegistrationDate': '2015-02-20', 'engineType': '5AM', 'engineRatio': '450', 'deliveryCountry': {'code': 'FR', 'label': 'FRANCE'}, 'family': {'code': 'X10', 'label': 'FAMILLE X10', 'group': '007'}, 'tcu': {'code': 'TCU0G1', 'label': 'TCU VER 0 GEN 1', 'group': 'E70'}, 'navigationAssistanceLevel': {'code': 'NAV3G3', 'label': 'AIDE NAVIG 3 NIVEAU 3', 'group': '408'}, 'battery': {'code': 'BT4MR1', 'label': 'BATTERIE BT4MR1', 'group': '968'}, 'radioType': {'code': 'RAD37A', 'label': 'RADIO 37A', 'group': '425'}, 'registrationCountry': {'code': 'FR'}, 'brand': {'label': 'RENAULT'}, 'model': {'code': 'X101VE', 'label': 'ZOE', 'group': '971'}, 'gearbox': {'code': 'BVEL', 'label': 'BOITE A VARIATEUR ELECTRIQUE', 'group': '427'}, 'version': {'code': 'INT MA 7'}, 'energy': {'code': 'ELEC', 'label': 'ELECTRIQUE', 'group': '019'}, 'registrationNumber': 'DP402ER', 'vcd': 'SYTINC/SKTPOU/SAND41/FDIU1/SSESM/MAPSTD/SSCALL/SAND88/SAND90/SQKDRO/SDIFPA/FACBA1/PRLEX1/SSRCAR/CABDO2/TCU0G1/SWALBO/STANDA/X10/B10/EA2/MA/ELEC/DG/TEMP/TR4X2/RV/ABS/CAREG/LAC/VT002/CPE/RET02/SPROJA/RALU16/CEAVRH/AIRBA1/SERIE/DRA/DRAP03/HARM01/ATAR/TEQNC/SFBANA/KM/SDPRPN/AVREPL/SSDECA/ASRESP/RDAR02/ALEVA/CACBL2/SOP02C/CTHAB2/TRNOR/LVAVIP/LVAREL/SASURV/KTGREP/SGSCHA/APL03/ALOUCC/CMAR3P/NAV3G3/RAD37A/BVEL/AUTNOR/RNORM/ISOFIX/EQPEUR/HRGM01/SDPCLV/TLFRAN/SPRODI/SAN613/SSAPEX/ELC1/SANCML/PE2012/PHAS1/SAN913/026KWH/BT4MR1/VEC010/X101VE/NBT007/5AM', 'assets': [{'assetType': 'PICTURE', 'renditions': [{'resolutionType': 'ONE_MYRENAULT_LARGE', 'url': 'https://3dv2.renault.com/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA2%2FDG%2FVT002%2FRET02%2FRALU16%2FDRAP03%2FHARM01%2FTEQNC%2FRDAR02%2FALEVA%2FSOP02C%2FTRNOR%2FLVAVIP%2FLVAREL%2FNAV3G3%2FRAD37A%2FSDPCLV%2FTLFRAN%2FSAN913%2FBT4MR1%2FNBT007&databaseId=1bc39d1c-ca60-4d83-9198-9bfc61d2da12&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_LARGE'}, {'resolutionType': 'ONE_MYRENAULT_SMALL', 'url': 'https://3dv2.renault.com/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA2%2FDG%2FVT002%2FRET02%2FRALU16%2FDRAP03%2FHARM01%2FTEQNC%2FRDAR02%2FALEVA%2FSOP02C%2FTRNOR%2FLVAVIP%2FLVAREL%2FNAV3G3%2FRAD37A%2FSDPCLV%2FTLFRAN%2FSAN913%2FBT4MR1%2FNBT007&databaseId=1bc39d1c-ca60-4d83-9198-9bfc61d2da12&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_SMALL_V2'}]}, {'assetType': 'PDF', 'assetRole': 'GUIDE', 'title': 'PDF Guide', 'description': '', 'renditions': [{'url': 'https://cdn.group.renault.com/ren/gb/myr/assets/x101ve/manual.pdf.asset.pdf/1558704861735.pdf'}]}, {'assetType': 'URL', 'assetRole': 'GUIDE', 'title': 'e-guide', 'description': '', 'renditions': [{'url': 'http://gb.e-guide.renault.com/eng/Zoe'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': '10 Fundamentals about getting the best out of your electric vehicle', 'description': '', 'renditions': [{'url': '39r6QEKcOM4'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Automatic Climate Control', 'description': '', 'renditions': [{'url': 'Va2FnZFo_GE'}]}, {'assetType': 'URL', 'assetRole': 'CAR', 'title': 'More videos', 'description': '', 'renditions': [{'url': 'https://www.youtube.com/watch?v=wfpCMkK1rKI'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Charging the battery', 'description': '', 'renditions': [{'url': 'RaEad8DjUJs'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Charging the battery at a station with a flap', 'description': '', 'renditions': [{'url': 'zJfd7fJWtr0'}]}], 'yearsOfMaintenance': 12, 'deliveryDate': '2015-02-20', 'retrievedFromDhs': False}}]}
[... cut ...]
DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {'data': {'type': 'Car', 'id': 'VF1AGVYA051xxxxxx', 'attributes': {'totalMileage': 58267}}}
Traceback (most recent call last):
  File "/usr/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.3.0', 'console_scripts', 'pyze')()
  File "/usr/lib/python3.8/site-packages/PyZE-0.3.0-py3.8.egg/pyze/cli/__main__.py", line 68, in main
  File "/usr/lib/python3.8/site-packages/PyZE-0.3.0-py3.8.egg/pyze/cli/status.py", line 64, in run
KeyError: 'hvacStatus'

jamesremuscat, 2019-12-17:
The Renault API is genuinely not returning `hvacStatus` in the `hvac-status` call. Well, there we go!

This'll be fixed by #21.

PR #21 (2019-12-17): Handle `hvacStatus` key missing from hvac status call.
Not sure quite how this information can be missing from the `hvac-status` API call, but apparently it can be...

Fixes #19.
commit ffdc1959 (2019-12-17): Handle `hvacStatus` key missing from hvac status call.

Not sure quite how this information can be missing from the
`hvac-status` API call, but apparently it can be...

Fixes #19.
closed 2019-12-17 by jamesremuscat