jamesremuscat/pyze #20: Access is denied for this resource

nvallas, 2019-12-17:
pyze status no longer work for few weeks.
Strangly debug output give me most of information I need

$ pyze login
Enter your My Renault email address: vallas@domain.fr
Enter your password: 
WARNING: Multiple Kamereon accounts found:
- 5508bd90…
- 722fc627…
- bb5a7861…
Using the first of these. If that's not correct (perhaps you can't see your vehicle)
or to silence this message, run `pyze set-account` or set the KAMEREON_ACCOUNT_ID
environment variable to the account you want to use i.e.
    KAMEREON_ACCOUNT_ID=abcdef123456789 pyze ...
API users may instead call Kamereon#set_account_id().
This setting will persist until you next log in.
Logged in successfully.

$ export KAMEREON_ACCOUNT_ID="5508bd90-…"
$ export KAMEREON_API_KEY="oF09W…"
$ export GIGYA_API_KEY="3_4LKbCcMM…"

$ pyze vehicles
Found 1 vehicle
 - XXXXXXX: RENAULT ZOE [VFXXXXXXXXXXXXX]

$  pyze  status
Error communicating with Renault API!
{"errors":[{"status":"403","code":"security.access","detail":"Access is denied for this resource"}]}

$  pyze --debug  status

WARNING:pyze:Debug output enabled. Logs may contain personally identifiable information and account credentials! Be sure to sanitise these logs before sending them to a third party or posting them online.
DEBUG:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): api-wired-prod-1-euw1.wrd-aws.com
DEBUG:requests.packages.urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/5508bd90-XXXXXXXXXXXXXXXXXXXXX/vehicles?country=FR HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicles response: {'vehicleLinks': [{'status': 'ACTIVE'
	, 'createdDate': '2019-10-06T11:15:45.468978Z'
	, 'mileage': 34001
	, 'brand': 'RENAULT'
	, 'connectedDriver': {'createdDate': '2019-10-17T07:00:16.063939Z'
	, 'lastModifiedDate': '2019-10-17T07:00:16.063939Z'
	, 'role': 'MAIN_DRIVER'}
	, 'startDate': '2019-10-06'
	, 'annualMileage': 10000
	, 'preferredDealer': {'createdDate': '2019-10-06T17:55:42.156569Z'
	, 'dealerId': 'XXXXXXXXXX'
	, 'brand': 'RENAULT'
	, 'lastModifiedDate': '2019-10-06T17:55:42.156569Z'}
	, 'vehicleDetails': {'family': {'label': 'FAMILLE X10'
	, 'code': 'X10'
	, 'group': '007'}
	, 'gearbox': {'label': 'BOITE A VARIATEUR ELECTRIQUE'
	, 'code': 'BVEL'
	, 'group': '427'}
	, 'brand': {'label': 'RENAULT'}
	, 'navigationAssistanceLevel': {'label': 'LEVEL 3 TYPE 5 NAVIGATION'
	, 'code': 'NAV3G5'
	, 'group': '408'}
	, 'assets': [{'assetType': 'PICTURE'
	, 'renditions': [{'url': 'https://3dv2.renault.com	
/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA1%2FDG%2FVT003%2FRET02%2FRTOL15%2FDRAP07%2FHARM01%2FOV369%2FSRDPRO%2FSALEVA%2FSOP01C%2FTRNOR%2FLVAVEL%2FLVARMA%2FNAV3G5%2FRAD35A%2FSDPCLV%2FTLFRAN%2FSAN913%2FBT4MR1%2FNBT007
	&databaseId=1bc39d1c-XXXXXXXXXXXXXXXXXXXXX
	&bookmarkSet=RSITE
	&bookmark=EXT_34_DESSUS
	&profile=HELIOS_OWNERSERVICES_LARGE'
	, 'resolutionType': 'ONE_MYRENAULT_LARGE'}
	, {'url': 'https://3dv2.renault.com/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA1%2FDG%2FVT003%2FRET02%2FRTOL15%2FDRAP07%2FHARM01%2FOV369%2FSRDPRO%2FSALEVA%2FSOP01C%2FTRNOR%2FLVAVEL%2FLVARMA%2FNAV3G5%2FRAD35A%2FSDPCLV%2FTLFRAN%2FSAN913%2FBT4MR1%2FNBT007
	&databaseId=1bc39d1c-XXXXXXXXXXXXXXXXXXXXX
	&bookmarkSet=RSITE
	&bookmark=EXT_34_DESSUS
	&profile=HELIOS_OWNERSERVICES_SMALL_V2'
	, 'resolutionType': 'ONE_MYRENAULT_SMALL'}]}]
	, 'registrationCountry': {'code': 'FR'}
	, 'deliveryCountry': {'label': 'FRANCE'
	, 'code': 'FR'}
	, 'model': {'label': 'ZOE'
	, 'code': 'X101VE'
	, 'group': '971'}
	, 'engineRatio': '601'
	, 'vin': 'VFXXXXXXXXXXXXX'
	, 'registrationNumber': 'XXXXXXX'
	, 'yearsOfMaintenance': 12
	, 'registrationDate': '2016-XX-XX'
	, 'tcu': {'label': 'TCU VER 0 GEN 2'
	, 'code': 'TCU0G2'
	, 'group': 'E70'}
	, 'engineType': '5AQ'
	, 'battery': {'label': 'BATTERIE BT4MR1'
	, 'code': 'BT4MR1'
	, 'group': '968'}
	, 'deliveryDate': '2016-XX-XX'
	, 'version': {'code': 'LIF MB 9'}
	, 'energy': {'label': 'ELECTRIQUE'
	, 'code': 'ELEC'
	, 'group': '019'}
	, 'vcd': 'SYTINC/SKTPOU/SAND41/FDIU1/SSESM/MAPSTD/SSCALL/SAND88/SAND90/SQKDRO/SDIFPA/FACBA1/PRLEX1/SSRCAR/CABDO2/TCU0G2/SWALBO/EVTEC1/STANDA/X10/B10/EA1/MB/ELEC/DG/TEMP/TR4X2/RV/ABS/CAREG/LAC/VT003/CPE/RET02/SPROJA/RTOL15/CEAVRH/AIRBA1/SERIE/DRA/DRAP07/HARM01/ATAR/OV369/SFBANA/KM/DPRPN/AVREPL/SSDECA/ASRESP/SRDPRO/SALEVA/CACBL2/SOP01C/CTHAB2/TRNOR/LVAVEL/LVARMA/SASURV/KTGREP/SGSCHA/APL03/ALOUCC/CMAR3P/NAV3G5/RAD35A/BVEL/AUTNOR/RNORM/ISOFIX/EQPEUR/HRGM01/SDPCLV/TLFRAN/SPRODI/SAN613/SSAPEX/ELC1/SANCML/PE2012/PHAS1/SAN913/026KWH/BT4MR1/VEC049/X101VE/NBT007/5AQ'
	, 'radioType': {'label': 'RADIO 35A'
	, 'code': 'RAD35A'
	, 'group': '425'}
	, 'retrievedFromDhs': False}
	, 'lastModifiedDate': '2019-10-17T07:00:16.064787Z'
	, 'linkType': 'USER'
	, 'mileageDate': '2019-10-16'
	, 'garageBrand': 'renault'
	, 'cancellationReason': {}
	, 'vin': 'VFXXXXXXXXXXXXX'
	, 'estimatedMileage': 32300}]
	, 'accountId': '5508bd90-XXXXXXXXXXXXXXXXXXXXX'
	, 'country': 'FR'}
DEBUG:requests.packages.urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VFXXXXXXXXXXXXX/battery-status HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {'data': {'id': 'VFXXXXXXXXXXXXX'
	, 'attributes': {'plugStatus': 0
	, 'chargeStatus': -1
	, 'batteryLevel': 47
	, 'batteryTemperature': 12
	, 'lastUpdateTime': '2019-12-17T08:14:28+01:00'
	, 'rangeHvacOff': 47}
	, 'type': 'Car'}}
DEBUG:requests.packages.urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VFXXXXXXXXXXXXX/charge-mode HTTP/1.1" 403 None

I look at code and debug infos but I don't understand how to correct this issue.

jamesremuscat, 2019-12-17:
Thanks for the detailed report.

I actually saw the same error myself a couple of weeks ago, and it cleared up by itself, so my working assumption is that it's Renault's API that's at fault.

Do you get the same error across all three of your account IDs?

nvallas, 2019-12-17:
No, I use environment variable like
$ export KAMEREON_ACCOUNT_ID="722fc627-…"
or
$ export KAMEREON_ACCOUNT_ID="bb5a7861-…"

and  I have this message 

$ pyze status
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.3.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.3.0-py3.5.egg/pyze/cli/__main__.py", line 68, in main
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.3.0-py3.5.egg/pyze/cli/status.py", line 19, in run
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.3.0-py3.5.egg/pyze/cli/common.py", line 41, in get_vehicle
RuntimeError: No vehicles found for this account!

jamesremuscat, 2020-01-06:
@nvallas Do you still encounter the same error?

As I said in https://github.com/jamesremuscat/pyze/issues/20#issuecomment-566516206, my suspicion is that this is a transient error on Renault's end, since there's no reason why the request to `/battery-status` should succeed and `/charge-mode` fail.

Unless (this is a wild guess)...

For some reason, Renault split access to the API into two packages, in the UK called "MY ZE Connect with navigation" and "MY ZE Inter@ctive". One's supposedly for read-only access, and one is supposed to also allow preconditioning and charge scheduling. Do you have both of these on your account, or only one? You can visit the R-Link store for your country and click "My services" to see them listed:

[image]

nabossha, 2020-01-08:
i have exactly the same issue, installed pyZE via pip3 today. i do not have the MY Z.E.Interactive package subscription anymore since i control the charging at home by controlling my wall-charger using openHab. 

my last debug-lines look like this:
````
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1XXXXXX/battery-status HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {'data': {'type': 'Car', 'id': 'VF1XXXXXX', 'attributes': {'lastUpdateTime': '2020-01-08T21:10:22+01:00', 'plugStatus': 0, 'rangeHvacOff': 79, 'batteryLevel': 66, 'batteryTemperature': 13, 'chargeStatus': -1}}}
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1XXXXXX/charge-mode HTTP/1.1" 403 None
Error communicating with Renault API!
{"errors":[{"status":"403","code":"security.access","detail":"Access is denied for this resource"}]}
````

nvallas, 2020-01-08:
> @nvallas Do you still encounter the same error?
> 
> As I said in [#20 (comment)](https://github.com/jamesremuscat/pyze/issues/20#issuecomment-566516206), my suspicion is that this is a transient error on Renault's end, since there's no reason why the request to `/battery-status` should succeed and `/charge-mode` fail.
> 
> Unless (this is a wild guess)...
> 
> For some reason, Renault split access to the API into two packages, in the UK called "MY ZE Connect with navigation" and "MY ZE Inter@ctive". One's supposedly for read-only access, and one is supposed to also allow preconditioning and charge scheduling. Do you have both of these on your account, or only one? You can visit the R-Link store for your country and click "My services" to see them listed:
> 
> [image]

your intution was good
my My Z.E. Connect with Navigation  seem to be ended
I renew subscription yesterday and I will wait 48 or 72 hours to try again.
But in fact renault rlink website have a  bug because I subscribed in novembre 2019 when I bought my car and it  should have lasted one year to novembre 2020.
I will keep you informed

PR #33 (2020-01-22): Handle various possible error reponses when collating data.
This PR allows the `pyze status` command to handle error responses from the API endpoints it calls and still try to return useful information to the user.

If an API call results in an error, it will handle the exception and list "unavailable" for the relevant data item.

This is necessary because Renault's API is flaky enough that any call might fail for no particular reason on any particular attempt - see #24 - and because some people only have one of the two ZE Services subscriptions meaning some endpoints are unavailable to them   [... cut]
closed 2020-01-22 by jamesremuscat