jamesremuscat/pyze #22: RuntimeError: Gigya returned error 400093: GSKeyBase is invalid

mfonville, 2019-12-23:
Hi,

I have set-up the latest master branch of pyze.
I have fetched my keys for `nl_NL` with `curl https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_nl_NL.json |grep Prod -A2`

With the environment variables set with these values I cannot login, I get the following error:
```
/home/maarten/.local/bin/pyze login
Enter your My Renault email address: my.email.address@is.here.com
Enter your password: 
Traceback (most recent call last):
  File "/home/maarten/.local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.3.0', 'console_scripts', 'pyze')()
  File "/home/maarten/.local/lib/python3.7/site-packages/PyZE-0.3.0-py3.7.egg/pyze/cli/__main__.py", line 68, in main
  File "/home/maarten/.local/lib/python3.7/site-packages/PyZE-0.3.0-py3.7.egg/pyze/cli/login.py", line 14, in run
  File "/home/maarten/.local/lib/python3.7/site-packages/PyZE-0.3.0-py3.7.egg/pyze/api/gigya.py", line 47, in login
  File "/home/maarten/.local/lib/python3.7/site-packages/PyZE-0.3.0-py3.7.egg/pyze/api/gigya.py", line 126, in raise_gigya_errors
RuntimeError: Gigya returned error 400093: GSKeyBase is invalid, no version: oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB
```

I have double-checked the Gigya key, and it is correct, could you help and figure out why it fails?

Thank you! (and for developing this api/cli!)

jamesremuscat, 2019-12-23:
Hi @mfonville,

Have you accidentally switched the Gigya and Kamereon API keys? The error is reported from Gigya but the `of09...` key it mentions is the Kamereon one.

mfonville, 2019-12-23:
Ah yes, that was the problem!

That means the Readme.md has a wrong instruction, it is reversed!

closed 2019-12-23 by mfonville