jamesremuscat/pyze #24: HTTP 502 error for hvac-status

mfonville, 2019-12-23:
When trying to run `pyze status`, I noticed that it would crash, so I ran with ``--debug`` and looking through what is going wrong.
And apparently for my car, retrieving `hvac-status` returns a HTTP 502 error:
```
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/XXXXXXXXX/hvac-status HTTP/1.1" 502 None
```
But the (CLI) code is not written to deal with this situation, it assumes is always available.
Is it unique to my car, that it is not available, did the resource location change, or should the CLI code just be updated to deal with resources that are not available?

mfonville, 2019-12-23:
Or is it maybe, because the car is not connected to a charging station when querying?
e.g. my `battery_status` just contains `{'chargeStatus': 0, 'batteryLevel': 68, 'plugStatus': 0}`

ArnoutVos, 2019-12-24:
`chargeStatus` is -1 when not charging and not plugged in at my car.

My full response reads: `attributes': {'lastUpdateTime': '2019-12-24T08:58:23+01:00', 'plugStatus': 0, 'rangeHvacOff': 73, 'batteryLevel': 56, 'batteryTemperature': 9, 'chargeStatus': -1}`

502 Bad Gateway means you've probably stumbled upon a bug on the AWS API side, maybe due to your car characteristics.

mfonville, 2019-12-24:
1. This is the first time I am trying pyze
2. The Renault app works perfectly fine
3. Yes the car is "Connected" when using it, but at the time it was parked and thus standby, so it was not really live I guess (and the Renault app works shows the data of earlier that day)

I have the new Zoe ZE50, maybe it has different characteristics? And if so, how could we check that and could I research the API methods?

ArnoutVos, 2019-12-24:
Ok, so this could be related to the new 2019 Zoe model. It's strange that lastUpdateTime is missing in the response because that's the timestamp also in use by the app at the top of the screen to show when it was last updated...

jamesremuscat, 2019-12-24:
Thanks for this report.

Not having `/hvac-status` available means that you'll be unable to retrieve the current preconditioning timer information from Renault. If you set a preconditioning timer in the app, does it show there?

It could just be a transient error on Renault's end - they tend to come and go, so it might magically start working in a couple of days' time.

> I have the new Zoe ZE50, maybe it has different characteristics? And if so, how could we check that and could I research the API methods?

I know that the ZE50 has some different API characteristics (I hear that the API couldn't cope with the ZE50 at all at its launch), but this is one I'd expect to remain the same. I also suspect that the `/location` and `/lock-status` endpoints might be ZE50-specific - or else why would they be there? It could be that they've only managed to cobble together the bare minimum of integration for the ZE50 so far, which is why your `battery-status` response is so sparse.

Perhaps I need to take a look at an updated copy of the app and see if there are any interesting changes there (that's how I discovered the first set of API endpoints).

None of which really helps us resolve this issue!

> should the CLI code just be updated to deal with resources that are not available?

I think this is the way forward. We should report e.g. `AC state: unavailable` if an endpoint returns an error, as it has here and in #20. I note too that the estimated range isn't included in your `battery-status` and that's another scenario we should cope with.

mfonville, 2019-12-24:
@jamesremuscat let me know how I can help you out with checking out any of the new API end points with the app. You can contact me privately if necessary.
I will check out those `/location` and `lock-status` endpoints if I find some time!

mfonville, 2019-12-31:
I am not sure what Renault is doing/changing in the backend, but when trying pyze today I now get this error/message:
```
pyze status
Error communicating with Renault API!
{"errors":[{"status":"502","code":"gateway.processing","title":"VNEXT","detail":"invalid vehicle state in order to be processed by gateway, invalid can : C1A"}]}
```
Not seen that before. Their app still works without any problem

PR #33 (2020-01-22): Handle various possible error reponses when collating data.
This PR allows the `pyze status` command to handle error responses from the API endpoints it calls and still try to return useful information to the user.

If an API call results in an error, it will handle the exception and list "unavailable" for the relevant data item.

This is necessary because Renault's API is flaky enough that any call might fail for no particular reason on any particular attempt - see #24 - and because some people only have one of the two ZE Services subscriptions meaning some endpoints are unavailable to them   [... cut]
closed 2020-01-22 by jamesremuscat