jamesremuscat/pyze #26: When req Token: Kamereon authentication failed! No token ID found

boekabart, 2020-01-03:
Login etc goes well, I can also get good responses for the persons and accounts API, listing my details and even car VIN... but when getting a Token, it fails. Digging down, this is the response ont the .../kamereon/token call:

```
{
    "type": "FUNCTIONAL",
    "messages": [{
        "code": "err.func.user.invalid",
        "message": "Kamereon authentication failed! No token ID found."
    }],
    "errors": [{
        "errorCode": "err.func.user.invalid",
        "errorMessage": "Kamereon authentication failed! No token ID found."
    }],
    "error_reference": "FUNCTIONAL"
}
```

Note that I did change the country parameter to `NL`, otherwise none of the APIs would work.

Any clues?

boekabart, 2020-01-03:
By the way, Feature request: if case of non-200 response codes, it would be great if the pyze / api would log the actual response body - I had to basically rebuild the API in Fiddler to get there...

ArnoutVos, 2020-01-03:
You can add the `pyze --debug` flag to display the response body.

jamesremuscat, 2020-01-06:
That's very odd. The call to list vehicles on an account itself requires a Kamereon token, so I'm not sure how that could be succeeding while the call to get a token itself is failing! I wonder if you managed to time the requests just wrong and the token expired mid-request? But presumably this is a repeatable error or else you wouldn't have needed to raise it here!

What's the exact sequence of `pyze` commands you're using to reproduce the error? As @ArnoutVos suggests, can you try running them with `--debug` and posting the results here (you'll have to edit out various tokens and credentials)?

I've also never noticed any effect from changing the `country` parameter; I can change from `GB` no `NL` or even `QQ` and everything just keeps working.

boekabart, 2020-01-06:
The call I make is simply:
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxx-xxx-xxx/vehicles?country=NL

With the correct `x-gigya-id_token` (from `g.get_jwt_token()`) and apiKey supplied, that works, and lists quite a bit of data about my car (configuration, type of "gearbox" etc), but not really EV related stuff.

This data also matches with the data I can see in the MyRenault app. Yes, it knows the car, image, type, license plate, VIN... but no detailed data like SoC, location or 'interaction' possible.

Does it maybe make sense to drop a (slightly censored) result of that call here, so that someone with a working account can compare notes?

boekabart, 2020-01-06:
```
{
    "accountId": "xxxx-xxxx",
    "country": "NL",
    "vehicleLinks": [{
        "brand": "RENAULT",
        "vin": "VF1AG0007XXXXXXXX",
        "status": "ACTIVE",
        "linkType": "USER",
        "garageBrand": "RENAULT",
        "startDate": "2020-01-02",
        "createdDate": "2020-01-02T16:55:54.002996Z",
        "lastModifiedDate": "2020-01-02T16:55:54.002996Z",
        "cancellationReason": {},
        "vehicleDetails": {
            "vin": "VF1AG0007XXXXXXXX",
            "registrationDate": "2019-12-19",
            "firstRegistrationDate": "2019-12-19",
            "engineType": "5AQ",
            "engineRatio": "605",
            "deliveryCountry": {
                "code": "NL",
                "label": "PAYS-BAS"
            },
            "family": {
                "code": "X10",
                "label": "FAMILLE X10",
                "group": "007"
            },
            "tcu": {
                "code": "AIVCT",
                "label": "AVEC BOITIER CONNECT AIVC",
                "group": "E70"
            },
            "navigationAssistanceLevel": {
                "code": "SAN408",
                "label": "CRITERE DE CONTEXTE",
                "group": "408"
            },
            "battery": {
                "code": "BT4AR1",
                "label": "BATTERIE BT4AR1",
                "group": "968"
            },
            "radioType": {
                "code": "NA419",
                "label": "FULL NAV DAB ETH - PREM",
                "group": "425"
            },
            "registrationCountry": {
                "code": "NL"
            },
            "brand": {
                "label": "RENAULT"
            },
            "model": {
                "code": "X102VE",
                "label": "ZOE",
                "group": "971"
            },
            "gearbox": {
                "code": "BVEL",
                "label": "BOITE A VARIATEUR ELECTRIQUE",
                "group": "427"
            },
            "version": {
                "code": "ED1 A MC 1LP"
            },
            "energy": {
                "code": "ELEC",
                "label": "ELECTRIQUE",
                "group": "019"
            },
            "registrationNumber": "G267AA",
            "vcd": "ASCOD0/DLIGM2/SSTINC/KITPOU/SKTPGR/SSCCPC/SDPSEC/FDIU2/SSMAP/SSCALL/FACBA1/DANGMO/SSRCAR/SSCABD/AIVCT/AVGSI/ITPK7/VOLCHA/REACTI/AVOSP1/SWALBO/SSDWGE/1234Y/SSAEBS/PRAHL/RRCAM/STANDA/X10/B10/EA3/MC/ELEC/DG/TEMP/TR4X2/AFURGE/RV/ABS/CAREG/LAC/VSTLAR/CPETIR/RET03/PROJAB/RALU16/CEAVRH/ADAC/AIRBA2/SERIE/CUI/CUIR02/HARM02/3ATRPH/SGAV02/BARRAB/TEGNE/SFBANA/KM/DPRPN/AVREPL/SSDECA/ABLAV/ASRESP/ALEVA/CACBL2/SOP02C/STHPLG/SKTGRV/VLCUI2/RETRCR/TRSEV1/RETC/LVAVIP/LVAREL/SASURV/KTGREP/SGACHA/NLD01/APL03/MONG01/FSTPO/ALOUC5/CMAR3P/SAN408/NA419/BVEL/AUTAUG/SPREST/RDIF01/PNSTRD/ISOFIX/EQPEUR/HRGM01/SDPCLV/CHARAP/TL01A/SPRODI/SAN613/AIRBDE/PSMREC/ELC1/SSPTLP/SANCML/SEXTIN/PE2019/PHAS2/EDI/THABT2/SSTYAD/SSHYB/052KWH/BT4AR1/VEC056/X102VE/NBT022/5AQ",
            "assets": [{
                "assetType": "PICTURE",
                "renditions": [{
                    "resolutionType": "ONE_MYRENAULT_LARGE",
                    "url": "https://3dv2.renault.com/ImageFromBookmark?configuration=DLIGM2%2FKITPOU%2FDANGMO%2FITPK7%2FVOLCHA%2FREACTI%2FSSAEBS%2FPRAHL%2FRRCAM%2FX10%2FB10%2FEA3%2FDG%2FCAREG%2FVSTLAR%2FCPETIR%2FRET03%2FPROJAB%2FRALU16%2FCUIR02%2F3ATRPH%2FTEGNE%2FALEVA%2FVLCUI2%2FRETRCR%2FRETC%2FLVAVIP%2FLVAREL%2FSGACHA%2FNA419%2FRDIF01%2FTL01A%2FEDI%2FNBT022&databaseId=f9c25d10-8571-4b2c-95dd-2be990af437a&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_LARGE"
                }, {
                    "resolutionType": "ONE_MYRENAULT_SMALL",
                    "url": "https://3dv2.renault.com/ImageFromBookmark?configuration=DLIGM2%2FKITPOU%2FDANGMO%2FITPK7%2FVOLCHA%2FREACTI%2FSSAEBS%2FPRAHL%2FRRCAM%2FX10%2FB10%2FEA3%2FDG%2FCAREG%2FVSTLAR%2FCPETIR%2FRET03%2FPROJAB%2FRALU16%2FCUIR02%2F3ATRPH%2FTEGNE%2FALEVA%2FVLCUI2%2FRETRCR%2FRETC%2FLVAVIP%2FLVAREL%2FSGACHA%2FNA419%2FRDIF01%2FTL01A%2FEDI%2FNBT022&databaseId=f9c25d10-8571-4b2c-95dd-2be990af437a&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_SMALL_V2"
                }]
            }],
            "yearsOfMaintenance": 12,
            "deliveryDate": "2019-12-19",
            "retrievedFromDhs": false
        }
    }]
}
```

jamesremuscat, 2020-01-06:
@boekabart Are you using the Pyze API or CLI? Please could you show me what you're doing in order to receive the error message from your original description?

Looks like the `/vehicles` endpoint no longer requires a Kamereon token (or it possibly never did in the first place), but that's only tangentially relevant to why you are unable to get such a token.

boekabart, 2020-01-06:
Tried both, the error comes the moment the token request is made, no matter what 'high level' call I'm trying to make that (pyze thinks) requires a Token. So repro for me is just `pyze --debug vehicles` for example:

```
WARNING:pyze:Debug output enabled. Logs may contain personally identifiable information and account credentials! Be sure to sanitise these logs before sending them to a third party or posting them online.
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): accounts.eu1.gigya.com
DEBUG:urllib3.connectionpool:https://accounts.eu1.gigya.com:443 "POST /accounts.getJWT HTTP/1.1" 200 950
DEBUG:pyze.api.gigya:Received Gigya getJWT response: {'callId': 'b4b312b35dc44c118755356d37<zzzzzz>', 'errorCode': 0, 'apiVersion': 2, 'statusCode': 200, 'statusReason': 'OK', 'time': '2020-01-06T20:38:34.756Z', 'ignoredFields': '', 'id_token': 'ey-<token-censored-by-me>-g'}
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api-wired-prod-1-euw1.wrd-aws.com
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/d35f4ef6-14f0-4daf-9e96-42a552632041/kamereon/token?country=NL HTTP/1.1" 400 None
Error communicating with Renault API!
{"type":"FUNCTIONAL","messages":[{"code":"err.func.user.invalid","message":"Kamereon authentication failed! No token ID found."}],"errors":[{"errorCode":"err.func.user.invalid","errorMessage":"Kamereon authentication failed! No token ID found."}],"error_reference":"FUNCTIONAL"}
```

jamesremuscat, 2020-01-06:
Just comparing your vehicles listing to mine, and mine has a section that yours doesn't.

```json
{
    "accountId": "xxxxxxxxxxxxxx",
    "country": "GB",
    "vehicleLinks": [{
        "brand": "RENAULT",
        "vin": "Vxxxxxxxxx0",
        "status": "ACTIVE",
        "linkType": "USER",
        "garageBrand": "renault",
        "startDate": "2019-07-04",
        "createdDate": "2019-07-04T18:24:21.998736Z",
        "lastModifiedDate": "2019-07-04T18:42:08.627642Z",
        "cancellationReason": {},
        "connectedDriver": {
            "role": "MAIN_DRIVER",
            "createdDate": "2019-07-04T18:42:08.626932Z",
            "lastModifiedDate": "2019-07-04T18:42:08.626932Z"
        },
        "vehicleDetails": {
            "vin": "Vxxxxxxxxx0",
            "registrationDate": "2018-03-05",
            "firstRegistrationDate": "2018-03-05",
            "engineType": "5AM",
            "engineRatio": "450",
            "deliveryCountry": {
                "code": "GB",
                "label": "GRANDE BRETAGNE"
            },
            "family": {
                "code": "X10",
                "label": "FAMILLE X10",
                "group": "007"
            },
            "tcu": {
                "code": "TCU0G2",
                "label": "TCU VER 0 GEN 2",
                "group": "E70"
            },
            "navigationAssistanceLevel": {
                "code": "NAV3G5",
                "label": "LEVEL 3 TYPE 5 NAVIGATION",
                "group": "408"
            },
            "battery": {
                "code": "BT4AR1",
                "label": "BATTERIE BT4AR1",
                "group": "968"
            },
            "radioType": {
                "code": "RAD06D",
                "label": "RADIO 06D",
                "group": "425"
            },
            "registrationCountry": {
                "code": "GB"
            },
            "brand": {
                "label": "RENAULT"
            },
            "model": {
                "code": "X101VE",
                "label": "ZOE",
                "group": "971"
            },
            "gearbox": {
                "code": "BVEL",
                "label": "BOITE A VARIATEUR ELECTRIQUE",
[... 64 lines cut ...]
                "title": "Charging the battery",
                "description": "",
                "renditions": [{
                    "url": "RaEad8DjUJs"
                }]
            }, {
                "assetType": "VIDEO",
                "assetRole": "CAR",
                "title": "Charging the battery at a station with a flap",
                "description": "",
                "renditions": [{
                    "url": "zJfd7fJWtr0"
                }]
            }],
            "yearsOfMaintenance": 12,
            "deliveryDate": "2018-03-05",
            "retrievedFromDhs": false
        }
    }]
}
```

Ignoring the "assets" at the end, I also have a `connectedDriver` that yours is missing. I wonder if that's the reason the Renault API is refusing to give you a Kamereon token? Or perhaps it's entirely unrelated. 

(Aside: I can change what's given as the top-level `country` attribute by changing the `country` URL parameter.)

boekabart, 2020-01-08:
When I click 'connection' ("verbinding") upon startup of the car, in the profile selection screen, this leads to some error. I suppose there may be smth wrong with my account / the car. Raised with Renault.

ArnoutVos, 2020-01-08:
Be prepared, I have had also connection issues at car level and Renault was already 2 months 'busy' with my ticket when I pulled the right fuse (resetting the connectivity module) and problems were solved right away by doing that.

boekabart, 2020-01-11:
Fortunately, I got a message from Renault telling me to finally press the connect button, and after all the pairing magic, the API now simply works... That is, SOME API's work. 
In both the API and the official Renault app, the following are missing:
* Mileage
* Location
* Anything to do with charging.

Interestingly enough, I can get the battery level (85%) in the API, but can't see it in the app (nor have any idea where it would show).

As a person who's lucky enough to also own a Tesla, the amount of information the Renault and its API/App provides, is quite abysmal compared to 'the big example'... Impressingly disappointing!

I hope the v2 APIs I see mentioned in other tickets will improve things a bit.

closed 2020-01-11 by boekabart