jamesremuscat/pyze #27: Support v2 charge scheduler API

jamesremuscat, 2020-01-06:
New endpoint (GET): `car-adapter/v1/cars/{vin}/charging-settings`.

Example response:
```json
{
    "data": {
        "type": "Car",
        "id": "xxxxxxxxxxxxxxxxx",
        "attributes": {
            "mode": "scheduled",
            "schedules": [{
                "id": 1,
                "activated": true,
                "monday": {
                    "startTime": "T12:00Z",
                    "duration": 15
                },
                "tuesday": {
                    "startTime": "T04:30Z",
                    "duration": 420
                },
                "wednesday": {
                    "startTime": "T22:30Z",
                    "duration": 420
                },
                "thursday": {
                    "startTime": "T22:00Z",
                    "duration": 420
                },
                "friday": {
                    "startTime": "T12:00Z",
                    "duration": 15
                },
                "saturday": {
                    "startTime": "T12:00Z",
                    "duration": 15
                },
                "sunday": {
                    "startTime": "T12:00Z",
                    "duration": 15
                }
            }]
        }
    }
}
```

Looks vaguely the same, except:
- There's now an array of schedules each with a single time period per day, rather than each day having an array of time periods
- Time format has changed to something ISO-like (it does make clear that schedules are in UTC though)

I'm assuming the POST endpoint at `car-adapter/v2/cars/{vin}/actions/charge-schedule` behaves in a somewhat similar manner to the existing one. (Note the `v2` in this URL, which does _not_ appear in the GET endpoint, though it probably should have done...)

boekabart, 2020-01-12:
With the car set to 'charge now', the schedules comes back an empty array.

jamesremuscat, 2020-01-13:
@boekabart That's not what I get:

```
>>> v._get('charge-mode')
{"data":{"type":"Car","id":"VF1xxxxxxxxxxx","attributes":{"chargeMode":"always_charging"}}}
{'chargeMode': 'always_charging'}

>>> v._get('charging-settings')
{"data":{"type":"Car","id":"VF1xxxxxxxxxxx","attributes":{"mode":"scheduled","schedules":[{"id":1,"activated":true,"monday":{"startTime":"T12:00Z","duration":15},"tuesday":{"startTime":"T12:00Z","duration":15},"wednesday":{"startTime":"T12:00Z","duration":15},"thursday":{"startTime":"T12:00Z","duration":15},"friday":{"startTime":"T12:00Z","duration":15},"saturday":{"startTime":"T12:00Z","duration":15},"sunday":{"startTime":"T04:30Z","duration":420}}]}}}
{'mode': 'scheduled', 'schedules': [{'id': 1, 'activated': True, 'monday': {'startTime': 'T12:00Z', 'duration': 15}, 'tuesday': {'startTime': 'T12:00Z', 'duration': 15}, 'wednesday': {'startTime': 'T12:00Z', 'duration': 15}, 'thursday': {'startTime': 'T12:00Z', 'duration': 15}, 'friday': {'startTime': 'T12:00Z', 'duration': 15}, 'saturday': {'startTime': 'T12:00Z', 'duration': 15}, 'sunday': {'startTime': 'T04:30Z', 'duration': 420}}]
```

Or do you mean something else by "charge now"?

boekabart, 2020-01-15:
Probably yet another artefact of the connected service not being completely 'live' yet - similar result as you now.

PR #44 (2020-02-18): Implement "phase two" functionality
This PR implements a number of "phase two" functions from the Renault API:

- "Start charging now" endpoint and CLI command. Resolves #29.
- Use v2 of the battery status endpoint. Resolves #28.
  - The `pyze status` command works in much the same way, but now uses structured values for charge/plug status. Sadly these values are inconsistent across <=ZE40 and ZE50.
- Use v2 of the charge scheduler endpoints. Resolves #27.
  - v2 of the "get schedule" endpoint is actually under `/v1` because Renault
  - `pyze schedule` behaves in almost exactly the same way, b [... cut]
closed 2020-02-18 by jamesremuscat