jamesremuscat/pyze #29: Support "start charging now" API endpoint

jamesremuscat, 2020-01-06:
This was a `v1` endpoint that I didn't find the first time around.

(POST) `car-adapter/v1/cars/{vin}/actions/charging-start`

Request body:
```json
{
    "data": {
        "type": "ChargingStart",
        "attributes": {
           "action": "start"
        }
    }
}
```

Example response:
```json
{
    "data": {
        "type": "ChargingStart",
        "id": "abcdef12-3456-7890-abcd-ef1234567890",
        "attributes": {
           "action": "start"
        }
    }
}
```

boekabart, 2020-01-11:
Leads to an HTTP 400 for my ZE50:

```
{
    "errors": [{
        "status": "Bad Request",
        "code": "400",
        "title": "Http message is not readable",
        "detail": "JSON parse error: Unrecognized field \"data\" (class com.acms.platform.caradapter.datamodel.resource.action.ChargingStartActionResource), not marked as ignorable; nested exception is com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field \"data\" (class com.acms.platform.caradapter.datamodel.resource.action.ChargingStartActionResource), not marked as ignorable (3 known properties: \"id\", \"startDateTime\", \"action\"])\n at [Source: java.io.PushbackInputStream@66d28465; line: 2, column: 14] (through reference chain: com.acms.platform.caradapter.datamodel.resource.action.ChargingStartActionResource[\"data\"])"
    }]
}
```

I use content-type: application/json in my request headers.

jamesremuscat, 2020-01-13:
> I use content-type: application/json in my request headers.

That'd be your problem. The API expects `Content-Type: application/vnd.api+json`.

boekabart, 2020-01-15:
My bad - I did look for it in the blog post but must have read over it.... CTRL-F of course quickly pointed out my mistake now ;)

commit 7910a4d6 (2020-02-17): Add "start charging now" endpoint.

Resolves #29.
PR #44 (2020-02-18): Implement "phase two" functionality
This PR implements a number of "phase two" functions from the Renault API:

- "Start charging now" endpoint and CLI command. Resolves #29.
- Use v2 of the battery status endpoint. Resolves #28.
  - The `pyze status` command works in much the same way, but now uses structured values for charge/plug status. Sadly these values are inconsistent across <=ZE40 and ZE50.
- Use v2 of the charge scheduler endpoints. Resolves #27.
  - v2 of the "get schedule" endpoint is actually under `/v1` because Renault
  - `pyze schedule` behaves in almost exactly the same way, b [... cut]
closed 2020-02-18 by jamesremuscat