jamesremuscat/pyze #30: Charge-stats gives 500 error

Cheaternl, 2020-01-10:
When doing a request for the charge-stats it gives me this error.

pi@raspberrypi:~/renault$ pyze charge-stats
Error communicating with Renault API!
{"errors":[{"status":"Internal Server Error","code":"500","title":"Unexpected error","detail":"java.time.ZonedDateTime cannot be cast to java.time.YearMonth"}]}

pi@raspberrypi:~/renault$ pyze charge-stats --period day
Error communicating with Renault API!
{"errors":[{"status":"Internal Server Error","code":"500","title":"Unexpected error","detail":"java.time.ZonedDateTime cannot be cast to java.time.LocalDate"}]}

I've done the request for charge-history via Postman, and didn't get the response I wanted.
It seems that the API is broken down for this call, but maybe you can verify if that's the case?

jamesremuscat, 2020-01-13:
Yes, looks like something has changed since I last investigated this endpoint (which has only ever returned an empty array for me, though others have had more success).

jamesremuscat, 2020-01-13:
OK, as far as I can tell this is broken on Renault's end. Changing the format of the timestamps passed to that endpoint gives:

```
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1xxxxxxxxxxx/charge-history?type=month&start=20191013&end=20191213 HTTP/1.1" 400 None
Error communicating with Renault API!
{"errors":[{"status":"400","code":"IfValue","detail":"Wrong start time pattern for type MONTH, expected yyyyMM","source":{"parameter":"start"}},{"status":"400","code":"IfValue","detail":"Wrong end time pattern for type MONTH, expected yyyyMM","source":{"parameter":"end"}}]}
```

so they're definitely asking for `yyyyMM`; but it doesn't look like they're handling those parameters correctly on their backend. My guess is they're trying to compare them to a `java.time.ZonedDateTime` that they've created, without properly converting one to a comparable type... but hey, they employ programmers to worry about that sort of thing.

Closing this issue since the bug lies within Renault's code.

closed 2020-01-13 by jamesremuscat

Cheaternl, 2020-01-15:
Ok tnx for looking into this. 
I tried MANY date format combinations upon requesting their endpoint and NONE gave me a valid response that is useful. Only errors and hints how to solve it (which didn't work out either).

At least it's now confirmed by someone with more knowledge than me :-)
Hope we get this endpoint back soon with useful information, because I really want to know when I have charged and how many Kwh's I've charged my ZOE with, because that gives me proper usage information about my car. I can now charge the ZOE for free, but those charge-locations don't give information (headless devices) and I don't want to record all the charge actions myself manually.