jamesremuscat/pyze #31: Possible 'status' call available?

Cheaternl, 2020-01-10:
I was just bored with API calls not working, and tested some endpoint names.
I saw that

`https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1AG000________/status`

gave me:
```
{
    "errors": [
        {
            "status": "Not Found",
            "code": "404",
            "title": "Incorrect vehicle",
            "detail": "Vehicle not found"
        }
    ]
}
```

which is different than other endpoint names, which will result in:
```
{
    "timestamp": "2020-01-10T22:53:48.327+0000",
    "status": 404,
    "error": "Not Found",
    "message": "No message available",
    "path": "/car-adapter/v1/cars/VF1AG000_________/status2"
}
```

Since I don't know how you figure out the endpoint names, I am curious about how you do that.
Are you using some kind of program like Wireshark, to see what kind of calls are done by the Renault app? Or do you have an API documentation which mention all the API calls?

boekabart, 2020-01-11:
Yes, I can reproduce the same result. `status` has a different response that any other 'word' you put there.

jamesremuscat, 2020-01-23:
I wonder if this is something asinine like using a wildcard to specify which routes get passed to the backend? 

Since there are both `battery-status` and `hvac-status`, perhaps someone configured `*status$` to get forwarded to the backend server. It's clearly a different system returning the 404 at `/status` than the one at `/status2`.

Cheaternl, 2020-01-23:
And how did you figure out the existing API endpoints?

jamesremuscat, 2020-01-24:
Mostly through examination of the app - I won't go into detail here, feel free to drop me an email (same username, at gmail dot com).

famdrive, 2020-03-01:
If you leave the end point away it throw this json:

`{"data":{"type":"Car","id":"VF1xxxxxxxx","attributes":{"gearbox":"AUTOMATIC","versionCode":"INT A MC 1L","region":"EU","radioType":"NA419","energy":"electric","modelName":"ZOE","modelCode":"X102VE","modelCodeDetail":"X102VE","carGateway":"AVN","registrationCountry":"DE ","vehicleId":4xxxxxx,"deliveryCountry":"DE ","canGeneration":"C1A","brand":"RENAULT","engineType":"5AQ","batteryCode":"BT4AR1","tcuCode":"AIVCT","familyCode":"X10","registrationNumber":"","pictureURL":"https://3dv2.renault.com/ImageFromBookmark?configuration=DLIGM2%2FKITPOU%2FDANGMO%2FITPK4%2FVOLCHA%2FREACTI%2FSSAEBS%2FPRAHL%2FRRCAM%2FX10%2FB10%2FEA3%2FDG%2FCAREG%2FVSTLAR%2FCPETIR%2FRET03%2FPROJAB%2FRALU16%2FDRAP13%2F3ATRPH%2FTEKQA%2FALEVA%2FVLCUI2%2FRETRCR%2FRETC%2FLVAVIP%2FLVAREL%2FSGACHA%2FNA419%2FRDIF01%2FTL09A%2FNBT022&databaseId=f9cxxxxxxxxxxxxxxxxxx&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_LARGE"}}}`

Tried several endpoints without success, such as:

tires-status
tire-status
tires-pressure
pneu-status
etc.

jamesremuscat, 2020-06-01:
I'm going to close this, since it doesn't look like there's anything useful for us at `/status` after all.

closed 2020-06-01 by jamesremuscat