jamesremuscat/pyze #34: Investigate alternative URLs

jamesremuscat, 2020-01-21:
Just discovered a bunch of alternative URLs for API endpoints.

Whereas previously, vehicle endpoints were at
```
/commerce/v1/accounts/kmr/remote-services/car-adapter/v<VERSION>/cars/<VIN>/<ENDPOINT>
```
it appears a very similar set of endpoints are available at
```
/commerce/v1/accounts/<ACCOUNTID>/kamereon/kca/car-adapter/v<VERSION>/cars/<VIN>/<ENDPOINT>
```
and it is the latter that are used by the more recent versions of the app.

It seems that the `country` parameter must be specified for all URLs in the latter form.

I don't know if this is significant, or just a waste of time.

closed 2020-02-18 by jamesremuscat

g-mocken, 2020-04-03:
As of today, the old style endpoints no longer work (for me).

neutrino85, 2020-04-03:
No longer work for me too. And very unstable since yesterday.

mfonville, 2020-08-12:
I just wanted to give a heads-up that Renault has changed from localized My Renault-apps (that differed per country) to a single app for every user in every country: https://play.google.com/store/apps/details?id=com.renault.myrenault.one.fr&authuser=0

Maybe they also changed some API stuff with this new release.

commit 3e4a0400 (2020-02-17): Use new-style URLs that include account UUIDs.

Closes #34 - everything seems to just work once it's wired in like this.
PR #44 (2020-02-18): Implement "phase two" functionality
This PR implements a number of "phase two" functions from the Renault API:

- "Start charging now" endpoint and CLI command. Resolves #29.
- Use v2 of the battery status endpoint. Resolves #28.
  - The `pyze status` command works in much the same way, but now uses structured values for charge/plug status. Sadly these values are inconsistent across <=ZE40 and ZE50.
- Use v2 of the charge scheduler endpoints. Resolves #27.
  - v2 of the "get schedule" endpoint is actually under `/v1` because Renault
  - `pyze schedule` behaves in almost exactly the same way, b [... cut]