jamesremuscat/pyze #35: Internal Server Error code:500 with stats (ac or charge)

sumoo13, 2020-01-24:
hi,
have you the same error 
when i test in cli :
pyze charge-stats or pyze ac-stats
i get an Internal Server Error code:500 !

jamesremuscat, 2020-01-24:
`ac-stats` works for me, but `charge-stats` is broken on Renault's end and has been for a little while.

What's the full error message you get from `ac-stats`?

sumoo13, 2020-01-24:
full message error
Error communicating with Renault API!
{"errors":[{"status":"Internal Server Error","code":"500","title":"Unexpected error","detail":"Invalid response from the upstream server (The request sent to the GDC is erroneous) ; 502 Bad Gateway"}]}

jamesremuscat, 2020-01-25:
Yes, I'm afraid that's a problem with Renault's servers. There's nothing that pyze can do about it.

closed 2020-01-25 by jamesremuscat