jamesremuscat/pyze #36: CLI output in JSON format

henriktorget, 2020-02-02:
It would be very nice to have the CLI output data as JSON! :smiley: 

Something like this: 
`pyze status -json`

This way i wont have to make a wrapper for the python library.
And i wont have to parse the `pyze status` output.
Instead i can just call the CLI and parse the JSON for profit. 

Any Python programmers here that can implement this?

ArnoutVos, 2020-02-02:
`pyze --debug status` gives JSON fragments, is that suitable?

henriktorget, 2020-02-03:
@ArnoutVos Thanks for the input! I did not see that in the code. Sounds like what I'm after. I'll check it out later today.

epenet, 2020-02-03:
You could also use the pyze API instead of the CLI if you want to wrap it...
By the way, is there is a reason why the CLI and the API code base are separate? Shouldn't/Couldn't the CLI just be a command line wrapper around the API?

jamesremuscat, 2020-02-03:
> It would be very nice to have the CLI output data as JSON! 
> 
> Something like this:
> `pyze status -json`

That's a nice idea! 

The disadvantage of using `--debug` is that, as its name suggests, it spits out a lot of other debugging information - though it does give you the raw data from the API, you'd still need to parse the output to extract it.

A dedicated `--json` flag would make your life a lot easier - and sounds like a "good first issue" for anyone wanting to contribute to the project :slightly_smiling_face:

jamesremuscat, 2020-02-03:
> By the way, is there is a reason why the CLI and the API code base are separate? Shouldn't/Couldn't the CLI just be a command line wrapper around the API?

Because it's useful to be able to run `pyze status` rather than `pyze battery-status && pyze cockpit && pyze hvac-status && pyze charge-mode`? (Note to others stumbling on this thread: those aren't real `pyze` commands...)

I'm not sure I understand your point here. I'd hardly call sibling modules within the same repository "separate" codebases, but separation of concerns (here, communicating with Renault vs collating, displaying and manipulating the data) is good programming practice.

If I'm misunderstanding you, please do feel free to raise a PR to show me what you mean! :slightly_smiling_face:

epenet, 2020-02-03:
Sorry - I've reviewed the code and API/CLI split is fine - ignore what I said earlier.

henriktorget, 2020-02-05:
I'm implementing it myself now :)

henriktorget, 2020-02-05:
Quick question. I'm not a python developer, so my workflow right now is: 

1. Make a working snippet for solving small problem in Jupyter. 
2. Pasting code from jupyter into source code.
3. `python3 setup.py`
4. `pyze login`
5. `pyze status --json`
6. Cry
7. Repeat

Is there a better workflow I can easily set up?

jamesremuscat, 2020-02-05:
If you run `python3 setup.py develop`, then Python will link the installed version of `pyze` to your checkout, and any changes will be automatically picked up, so you don't have to keep running `setup.py`.

You shouldn't need to run `pyze login` more than once, as your credentials are stored in a local file.

nheinemans, 2020-08-06:
Just curious if there's any progress on this. I'm currently using the pyze status output and try to filter the output in order to push it in a TSDB (InfluxDB in my case). Getting the output from pyze directly in json format would make this process much easier and especially better suitable for future changes in the API.