jamesremuscat/pyze #38: KeyError: 'chargeStartDate'

karad0, 2020-02-08:
Hello,
Great Job. I integrate try to integrate to jeedom and homebridge.
But I have a problem with charge-history

> {'charges': [{'chargeStartDate': '2020-02-08T00:04:52Z', 'chargeEndDate': '2020-02-08T03:22:25Z', 'chargeDuration': 197, 'chargeStartBatteryLevel': 65, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 35, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2400, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-05T15:46:07Z', 'chargeEndDate': '2020-02-05T16:38:50Z', 'chargeDuration': 52, 'chargeStartBatteryLevel': 25, 'chargeEndBatteryLevel': 92, 'chargeBatteryLevelRecovered': 67, 'chargePower': 'standard', 'chargeStartInstantaneousPower': 19700, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-05T11:46:15Z', 'chargeEndDate': '2020-02-05T12:36:38Z', 'chargeDuration': 50, 'chargeStartBatteryLevel': 96, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 4, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2300, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-05T11:12:28Z', 'chargeEndDate': '2020-02-05T11:37:12Z', 'chargeDuration': 24, 'chargeStartBatteryLevel': 94, 'chargeEndBatteryLevel': 98, 'chargeBatteryLevelRecovered': 4, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2300, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-05T00:04:51Z', 'chargeEndDate': '2020-02-05T03:21:06Z', 'chargeDuration': 196, 'chargeStartBatteryLevel': 65, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 35, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2400, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-03T17:01:58Z', 'chargeEndDate': '2020-02-03T19:49:28Z', 'chargeDuration': 167, 'chargeStartBatteryLevel': 70, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 30, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2300, 'chargeEndStatus': 'ok'}, {'chargeStartDate': '2020-02-02T16:03:52Z', 'chargeEndDate': '2020-02-02T16:30:28Z', 'chargeDuration': 26, 'chargeStartBatteryLevel': 99, 'chargeEndBatteryLevel': 100, 'chargeBatteryLevelRecovered': 1, 'chargePower': 'slow', 'chargeStartInstantaneousPower': 2400, 'chargeEndStatus': 'ok'}, {'chargeEndDate': '2020-02-02T08:07:52Z', 'chargeEndBatteryLevel': 99, 'chargeEndStatus': 'error'}]}}}
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE', 'console_scripts', 'pyze')()
  File "/home/pi/pyze-develop/src/pyze/cli/__main__.py", line 68, in main
    parsed_args.func(parsed_args)
  File "/home/pi/pyze-develop/src/pyze/cli/charge-history.py", line 36, in run
    [_format_charge_history(h) for h in v.charge_history(from_date, to_date)],
  File "/home/pi/pyze-develop/src/pyze/cli/charge-history.py", line 36, in <listcomp>
    [_format_charge_history(h) for h in v.charge_history(from_date, to_date)],
  File "/home/pi/pyze-develop/src/pyze/cli/charge-history.py", line 54, in _format_charge_history
    ch['chargeStartDate']
KeyError: 'chargeStartDate'

jamesremuscat, 2020-02-14:
Because of course in Renault's world something can end without having started...!

Thanks for the bug report, I'll put a fix together.

closed 2020-02-14 by jamesremuscat

karad0, 2020-02-14:
It's blank
[image]

karad0, 2020-02-14:
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/XXXXXXXXXXXX/charges?start=20200201&end=20200214 HTTP/1.1" 200 None

jamesremuscat, 2020-02-14:
Correct, it's blank.

(That debug line wasn't the most useful one, but it does show that pyze is making the request to Renault's API correctly, and that the API is responding. I can't fix Renault's problems with that endpoint, though.)

karad0, 2020-02-14:
Ok thanks

PR #41 (2020-02-14): Handle more missing information in charge history response.
Fixes #38.
commit c1e341cf (2020-02-14): Handle more missing information in charge history response.

Fixes #38.