jamesremuscat/pyze #40: ZE50 Endpoint error

sumoo13, 2020-02-10:
not really an issue, just an information
I noticed that there were some errors with hvac endpoint for a ze50

> /hvac-status?country=FR): failed to open stream: HTTP request failed! HTTP/1.1 502 Bad Gateway
and charge endpoint

> charges?start=20200101&end=20200210&country=FR): failed to open stream: HTTP request failed! HTTP/1.1 403 Forbidden

some people with ze50 are the same error ?

neutrino85, 2020-02-10:
Same error here.

jamesremuscat, 2020-02-19:
Thanks for the info.

The first of those looks like a transient error on Renault's side; it might sort itself out in time.

As for the second: if that doesn't go away, be sure you have _both_ ZE Services subscriptions active on your account. That's an endpoint that Renault seem to be tinkering with quite a lot, so it's worth trying again later. (I seem to be saying that a lot with Renault's API...)

closed 2020-04-24 by jamesremuscat