jamesremuscat/pyze #43: New "battery inhibitor status" endpoint

jamesremuscat, 2020-02-17:
(GET): `/commerce/v1/accounts/{id}/kamereon/kca/car-adapter/v1/cars/{vin}/battery-inhibition-status`

**Does anyone get anything other than an HTTP `403` for this URL?**

I'm presuming it describes the setting that RCI use to prevent charging if a battery lease isn't fully paid up. Would be interested to see if battery-owned and battery-leased cars get different behaviour on this URL.

Should have a response roughly of the form:

```json
{
  "vin": "xxxxxxx",
  "lastUpdateTime": 1234567890,
  "batteryInhibitorStatus": 0
}
```

Values of `batteryInhibitorStatus` should be:
 - `0` - `CHARGE_BLOCKED`
- `1` - `NEXT_CHARGE_BLOCKED`
- All other values or `undefined`: `CHARGE_NOT_BLOCKED`

CalvinWalzel, 2020-02-18:
`/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/{vin}/battery-inhibition-status` also works and returns a `403`.

robertaramar, 2020-02-18:
I also get a 403

jamesremuscat, 2020-02-19:
@CalvinWalzel @robertaramar Are you battery-leased or battery-owned?

CalvinWalzel, 2020-02-19:
> @CalvinWalzel @robertaramar Are you battery-leased or battery-owned?

battery-owned

sumoo13, 2020-02-19:
with my car (ze40 battery-leased) it's return a 403 error code
with a ze50 battery-owned (not sure) [batteryInhibitorStatus] => 2

robertaramar, 2020-02-19:
My car is leased, so I assume the same goes for the battery.

famdrive, 2020-02-23:
Hi

I get

{"data":{"type":"Car","id":"VFxxxxxxxxxxx","attributes":{"lastUpdateTime":"2020-02-22T13:18:16Z","batteryInhibitorStatus":2}}}

I have a ZE50 leased also battery leased.

BTW, does anyone know how to read the range kilometers ?

KR famdrive

ArnoutVos, 2020-02-23:
> BTW, does anyone know how to read the range kilometers ?

[image]

The range is calculated on the remaining battery charge and the power usage based on your driving pattern. It's the same value as displayed on the left side of the dashboard.

[image]

famdrive, 2020-02-23:
Thank you for this information.
Where did you get this graphic from ? Do you know a formula for this ?

Dave-09, 2020-02-24:
If I see correctly, is that in Grafana? do you have the formula for me :-)

ArnoutVos, 2020-02-24:
> Thank you for this information.
> Where did you get this graphic from ? Do you know a formula for this ?

I'm running `pyze status` by a cronjob every 5 minutes and push changed data to InlfuxDB. Then Grafana shows the data in a nice graph.

jamesremuscat, 2020-03-03:
OK, so it looks like there's no particular pattern to whether or not this endpoint is available. (Pretty standard for this API sadly.)

We should nevertheless include the endpoint in pyze.