jamesremuscat/pyze #45: location does work for me

robertaramar, 2020-02-18:
you write in your WIKI https://github.com/jamesremuscat/pyze/wiki/Known-issues-with-the-Renault-API at the very bottom, that location does not work. It does work for me:

https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VF1AG000*********/location
    {
        "data": {
            "type": "Car",
            "id": "VF1AG000*********",
            "attributes": {
                "gpsLatitude": 48.*******,
                "gpsLongitude": 11.*******,
                "lastUpdateTime": "2020-02-18T16:58:38Z"
            }
        }
    }

jamesremuscat, 2020-02-19:
Thanks for letting me know - that's really useful. You should feel special, you're the first person who's got that URL to work :slightly_smiling_face:

closed 2020-02-19 by jamesremuscat

matthew1471, 2020-04-09:
@robertaramar is that by any chance with a ZE50 or a ZE40?

robertaramar, 2020-04-09:
Yes, it's a ZE50 bought this year

Dave-09, 2020-06-25:
This one works with ZE50, but not with ZE40 :)

lovetibber, 2020-10-21:
Hey @robertaramar @matthew1471 @Dave-09, does any of you have a clear view on which Zoe's support location and which doesn't? From the thread it looks like this is new in 2019's, but only for ZE50?

edddeduck, 2020-10-21:
Having owned a ZE20 ZE40 & ZE50 (and used Pyze on all 3) I can confirm the only one that provides a valid response to the location is the ZE50 model. I highly doubt they'll retrofit the feature to the older cars.

epenet, 2020-10-21:
Check this discussion: #83 

model X101VE doesn't return location data (returns error)
model X102VE returns location data, but only if the car actually has a GPS (returns empty object if car doesn't have a GPS)

I haven't (yet) worked out a way to distinguish between ZE50 with GPS and SE50 without GPS

lovetibber, 2020-10-21:
Thanks @epenet @edddeduck , much appreciated!