jamesremuscat/pyze #48: charge-schedule endpoint

neutrino85, 2020-02-21:
Not an issue, maybe an improvement.
I try to use the /actions/charge-schedule endpoint.
('/commerce/v1/accounts/kmr/remote-services/car-adapter/v2/cars/{vin}/actions/charge-schedule')
It requires an id and schedules.
I try to post 
`{'data' : {
        'type': 'ChargeSchedule',
        'attributes': {
             'id': 2,
             'schedules': {'monday': [{'startTime': 'T23:45Z', 'duration': 405}], 'tuesday': [{'startTime': 'T23:45Z', 'duration': 405}], 'wednesday': [{'startTime': 'T23:45Z', 'duration': 405}], 'thursday': [{'startTime': 'T23:45Z', 'duration': 405}], 'friday': [{'startTime': 'T23:45Z', 'duration': 405}], 'saturday': [{'startTime': 'T23:45Z', 'duration': 405}], 'sunday': [{'startTime': 'T23:45Z', 'duration': 405}]}
             }
    }}`
but it returns 
`{'errors': [{'status': 'Bad Request', 'code': '400', 'title': 'Http message is not readable', 'detail': 'com.fasterxml.jackson.databind.JsonMappingException: Can not deserialize instance of java.util.ArrayList out of START_OBJECT token\n at [Source: N/A; line: -1, column: -1] (through reference chain: com.acms.platform.caradapter.datamodel.status.charge.v2.ChargingSchedules["schedules"]); nested exception is com.fasterxml.jackson.databind.JsonMappingException: Can not deserialize instance of java.util.ArrayList out of START_OBJECT token\n at [Source: N/A; line: -1, column: -1] (through reference chain: com.acms.platform.caradapter.datamodel.status.charge.v2.ChargingSchedules["schedules"])'}]}}
`
I don't know how to format the schedules.

neutrino85, 2020-02-21:
Nevermind, I just try 
`{'type': 'ChargeSchedule', 'attributes': {'schedules': [{'id': 1, 'activated': True, 'monday': {'startTime': 'T23:45Z', 'duration': 405}, 'tuesday': {'startTime': 'T23:45Z', 'duration': 405}, 'wednesday': {'startTime': 'T23:45Z', 'duration': 405}, 'thursday': {'startTime': 'T23:45Z', 'duration': 405}, 'friday': {'startTime': 'T23:45Z', 'duration': 405}, 'saturday': {'startTime': 'T23:45Z', 'duration': 405}, 'sunday': {'startTime': 'T23:45Z', 'duration': 405}}, {'id': 2, 'activated': False}, {'id': 3, 'activated': False}, {'id': 4, 'activated': False}, {'id': 5, 'activated': False}]}}`
and it works :)

jamesremuscat, 2020-02-21:
Yes, Renault changed the data format for v2 of their API. You should find that it no longer requires all seven days, too!

pyze has supported the new format since `0.4.0` (released a couple of days ago).

closed 2020-02-21 by jamesremuscat