jamesremuscat/pyze #51: Support pushing the data to ABRP

mfonville, 2020-03-26:
ABRP (abetterrouteplanner.com) does allow nowadays to synchronize live data of the car to their servers for use in their app.
E.g. the battery state, projected range, charging state help for better real-time route planning and guidance.

Would it be possible to extend pyze that it can export/push data to ABRP?
Somebody already made a small hack to synchronize the information from their "Mijn KIA"-app to ABRP (using the Torque Pro API endpoint): https://github.com/TheBazeman/MijnKia

mfonville, 2020-05-07:
Just to give a small heads-up, the current pyze implementation didn't fit the requirements to write a good working integration with ABRP, so I have written a Zoe Dashboard that includes the integration.
It is still in testing phase, anyone who would like to participate in testing, please send me a message (you can probably find my email address on the internet quite easy :-) )

jamesremuscat, 2020-06-01:
I'm not sure what the action on pyze is here. Is there missing API that means you can't write a tool to export to ABRP?

mfonville, 2020-06-01:
One of the issues I did experience is that kamereon and gigya are coupled very tightly in PyZE and that the API assumes a bit too much that there is a single user (e.g. with the credentials storage, like another user tried to adjust with https://github.com/jamesremuscat/pyze/pull/62)

E.g. for my Dashboard and ABRP integration I have to store credentials in a database.

When done tinkering a bit more around with Renault's API for longer I hope to be able to propose/contribute some possible improvements.

Thank you for all your work on PyZE and exploring the API endpoints :-)

Sleevezipper87, 2020-08-28:
Hi @mfonville!

Did you ever get any further with your integration? If you're still on it, i'd love to help with development or testing!

mfonville, 2020-08-28:
@Sleevezipper87 yes, I do have a beta version at the moment. You can sign up at https://myrenau.lt/accounts/register/ with your MyRenault account. And use the app at at https://myrenau.lt/manager . You can enable ABRP-integration at the Vehicle-page.

It is sometimes still unstable though, it is still in development, and the UI is temporary.

mfonville, 2020-08-31:
I see there is currently a big with signing up new accounts, it will be resolved soon

cad0p, 2020-09-01:
> Hi @mfonville!
> 
> Did you ever get any further with your integration? If you're still on it, i'd love to help with development or testing!

me too, @mfonville great work!

CalvinWalzel, 2020-09-07:
@mfonville It's a really cool project. Does it refresh the data every 30 minutes or did you find a way around that limitation?

Also, maybe you could add an email or contact form to the UI in your project? That way, users can contact you for feedback.

mfonville, 2020-09-07:
@CalvinWalzel thanks!

Data is refreshed on a variable interval.
For telemetry, if a car is stale or a while, data is refreshed every 5 minutes. If a car has recently been active, it is updated every minute, if a car is active, it tries to update every 30 seconds (but ZE50 only reports every minute when driving, and old Zoe only when parked).
Location data is updated every 10 minutes (ZE50 only sends it when parked)

I will soon add some contact info and other stuff on the project page. It is still for a little small group of users (some friends, and the people who read PyZE issues comments ;-) )
There are still some showstopper bugs that I want to fix before at least announcing a beta publicly.