jamesremuscat/pyze #53: Authentication has changed

MattCheetham, 2020-04-04:
Looks like something has changed today.

I can log in fine but requesting any data such as the battery status endpoint is giving me

`{
    "error": {
        "title": "authentication",
        "code": "forbidden",
        "details": "authentication failed"
    }
}`

My live apps have stopped working but the My Renault app continues to function. I can't tell what's changed that I need to send to fix this.

jamesremuscat, 2020-04-04:
Are you using the old-style or new-style API endpoints, as described on #34?

A couple of people have reported that the old-style ones, without the account ID in the URL, are no longer working. pyze introduced support for the new-style URLs (PR #44) in version `0.4.0`.

MattCheetham, 2020-04-05:
Thank you. I thought I'd successfully updated to the new endpoints but had made a mistake missing one of the parameters out which was working fine until now but now seems to have stopped working. 

Matched the one's in your PRs and now we're all good again I think. Thanks for the help, I should have looked closer before opening this!

closed 2020-04-05 by MattCheetham