jamesremuscat/pyze #54: Requesting API-keys into pyze

Podariss, 2020-04-06:
Is it possible, for the ease of use, to incorporate the getting of the api-keys into pyze automaticly? As they seem to be the same for everybody?

"export KAMEREON_API_KEY=oF09WnKqvBDcr...
export GIGYA_API_KEY=3_e8d4g4SE_Fo8ahyH...
"

jamesremuscat, 2020-04-24:
At first I shied away from this to try and protect pyze, but now that time has passed I think this is a good idea.

mfonville, 2020-06-01:
The API keys differ per locale, and (might?) influence the features exposed in the API. Because the template with values imply that some features are country-related.
Btw, for who is interested, I had automatically fetched country/language data for all common ISO-codes and have listed the URLs of the "My Renault"-app per country (Android and iPhone) are available.

Attached you can find a fixture that I use for my dashboard application:
```
[
  {
    "model": "manager.countryconfig",
    "pk": 1,
    "fields": {
      "country_id": 1,
      "country_string": "bg_BG",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3__3ER_6lFvXEXHTP_faLtq6eEdbKDXd9F5GoKwzRyZq37ZQ-db7mXcLzR1Jtls5sn",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.bg"
    }
  },
  {
    "model": "manager.countryconfig",
    "pk": 2,
    "fields": {
      "country_id": 2,
      "country_string": "cs_CZ",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3_oRlKr5PCVL_sPWUZdJ8c5NOl5Ej8nIZw7VKG7S9Rg36UkDszFzfHfxCaUAUU5or2",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB"
    }
  },
  {
    "model": "manager.countryconfig",
    "pk": 3,
    "fields": {
      "country_id": 3,
      "country_string": "da_DK",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3_5x-2C8b1R4MJPQXkwTPdIqgBpcw653Dakw_ZaEneQRkTBdg9UW9Qg_5G-tMNrTMc",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.dk",
      "iphone_app": "dk/app/my-renault-danmark/id1409727390"
    }
  },
  {
    "model": "manager.countryconfig",
    "pk": 4,
    "fields": {
      "country_id": 4,
      "country_string": "de_DE",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.de",
      "iphone_app": "de/app/my-renault-deutschland/id1450827403"
    }
  },
  {
    "model": "manager.countryconfig",
    "pk": 5,
    "fields": {
      "country_id": 5,
      "country_string": "de_AT",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3__B4KghyeUb0GlpU62ZXKrjSfb7CPzwBS368wioftJUL5qXE0Z_sSy0rX69klXuHy",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.at",
[... 306 lines cut ...]
      "country_string": "sl_SI",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3_QKt0ADYxIhgcje4F3fj9oVidHsx3JIIk-GThhdyMMQi8AJR0QoHdA62YArVjbZCt",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.si",
      "iphone_app": "si/app/my-renault-slovenija/id1445022006"
    }
  },
  {
    "model": "manager.countryconfig",
    "pk": 29,
    "fields": {
      "country_id": 29,
      "country_string": "sv_SE",
      "gigya_url": "https://accounts.eu1.gigya.com",
      "gigya_api_key": "3_EN5Hcnwanu9_Dqot1v1Aky1YelT5QqG4TxveO0EgKFWZYu03WkeB9FKuKKIWUXIS",
      "kamereon_url": "https://api-wired-prod-1-euw1.wrd-aws.com",
      "kamereon_api_key": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB",
      "android_app": "com.renault.myrenault.one.se",
      "iphone_app": "se/app/my-renault-sverige/id1409727175"
    }
  }
]
```