jamesremuscat/pyze #55: api url change again

nvallas, 2020-04-23:
since Friday april 3 2020 api url changed from:

/commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/{vin}/battery-status

to

/commerce/v1/accounts/{id}/kamereon/kca/car-adapter/v2/cars/{vin}/battery-status

I'm trying to modify code to offer a PR

mfonville, 2020-04-23:
I believe this change has already been applied in January: https://github.com/jamesremuscat/pyze/commit/4cf5d66704dab8cabcbfd05a904b9e81a6cb62ac

nvallas, 2020-04-23:
My emoncms graph stopped in april 3
debug mode tell me 

DEBUG:requests.packages.urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/kmr/remote-services/car-adapter/v1/cars/VFXXXXXX123456789/battery-status HTTP/1.1" 404 None

I unpacked myrenault APK and I can see url difference between V4.1.0 38.4 (2020-04-04) and V2.6.12 (30.2 2019-06-22)

jamesremuscat, 2020-04-24:
As @mfonville suggests, pyze already uses the v2 battery status endpoint, and has done since version `0.4.0` released on the 19th February. Please make sure you're using an up-to-date version of pyze.

closed 2020-04-24 by jamesremuscat

nvallas, 2020-04-24:
You are totaly right
I did a mistake using bad version.
Sorry to disturbed you.