jamesremuscat/pyze #56: New "engine status" endpoint

jamesremuscat, 2020-04-24:
`/commerce/v1/accounts/{id}/kamereon/kca/car-adapter/v1/cars/{vin}/res-state` (`GET`)

Theoretically returns an object with `code` and `details`, where `code` is one of:

```
STOPPED_READY_FOR_KEY(11),
STOPPED_READY_FOR_RES(10),
RUNNING_BY_KEY(21),
RUNNING_BY_RES(42),
NOT_AVAILABLE(1),
INTERNAL_ERROR(0),
NOT_ENABLED(2),
NOT_YET_READY(3),
ERROR_CODE(9223372036854775808)
```

I don't know what "RES" is meant to mean.

Trying this endpoint on my ZE40 results in an `HTTP 403`; might be a ZE50-only thing, might be as inconsistent as some of the other endpoints, might not even be relevant to the Zoe (though the use of "engine" might just be force of habit).

Anyone else get anything useful here?

premultiply, 2020-04-24:
Maybe "RES" means "reservation" (for car sharing?). Who knows...

jamesremuscat, 2020-04-24:
Ah, think I figured it out: **R**emote **E**ngine **S**tart.

neutrino85, 2020-04-24:
On ze50 :
{'type': 'FUNCTIONAL', 'messages': [{'code': 'err.func.403', 'message':
'{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access
is denied for this resource\"}]}'}], 'errors': [{'errorCode':
'err.func.403', 'errorMessage':
'{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access
is denied for this resource\"}]}'}], 'error_reference': 'FUNCTIONAL'}

premultiply, 2020-04-24:
Remote Engine Start :-)
Hehe, a little bit useless on electric car, isn't it?
Maybe for remote sound module start :-D

epenet, 2020-04-24:
Has anyone tried the app or the pyze module on a non-electric recent Renault model?
I know pyze is directed at the the ZE models, but maybe Renault has it planned for non-ZE models...

mfonville, 2020-11-28:
In Renault's own Kamereon resource they use it for `getEngineStatus`

jumpjack, 2021-10-08:
I did setup an user friendly page for easy testing:

http://jumpjack.altervista.org/myrenault

On my Plugin Captur I get a "forbidden"  error too for res-status (and for actions/engine-start)