jamesremuscat/pyze #65: logged in but all properties unavailable

OhSoGood, 2020-05-17:
Hi,
First time I use your script (thank you for it!). I'm logged in, I see my vehicle (I had to call ' pyze set-account' as I had two Kamereon accounts found), 'pyze vehicles' return my vehicle model with its VIN, but when I do a 'pyze status', here's what I get:
--------------------  -------------
Battery level         Unavailable%
Range estimate        Unavailable
Plug state            Not available
Charging state        Not available
Charge mode           Unavailable
AC start at           Unavailable
External temperature  Unavailable
Total mileage         Unavailable
Updated at            Unavailable
--------------------  -------------

What could be wrong?

epenet, 2020-05-17:
You can try to run it with extra debug information:
> pyze status --debug

OhSoGood, 2020-05-17:
Here it is (pyze --debug status):

WARNING:pyze:Debug output enabled. Logs may contain personally identifiable information and account credentials! Be sure to sanitise these logs before sending them to a third party or posting them online.
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api-wired-prod-1-euw1.wrd-aws.com:443
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/12345678-3075-4da9-a711-8d49d69a6640/vehicles?country=GB HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicles response: {'accountId': '12345678-3075-4da9-a711-8d49d69a6640', 'country': 'GB', 'vehicleLinks': [{'brand': 'RENAULT', 'vin': 'VFxxxxxx', 'status': 'ACTIVE', 'linkType': 'OWNER', 'garageBrand': 'RENAULT', 'annualMileage': 16000, 'mileage': 32000, 'mileageDate': '2019-07-16', 'estimatedMileage': 18590, 'startDate': '2017-02-23', 'createdDate': '2019-05-23T20:45:16.598753Z', 'lastModifiedDate': '2020-05-17T16:50:42.343106Z', 'ownershipStartDate': '2017-01-31', 'cancellationReason': {}, 'preferredDealer': {'dealerId': '25002159_001', 'dealerName': 'GARAGE DU MOULIN', 'brand': 'RENAULT', 'createdDate': '2020-05-17T16:50:42.342665Z', 'lastModifiedDate': '2020-05-17T16:50:42.342665Z'}, 'vehicleDetails': {'vin': 'VFxxxxxx', 'registrationDate': '2017-01-01', 'firstRegistrationDate': '2017-01-01', 'engineType': '5AQ', 'engineRatio': '601', 'deliveryCountry': {'code': 'FR', 'label': 'FRANCE'}, 'family': {'code': 'X10', 'label': 'FAMILLE X10', 'group': '007'}, 'tcu': {'code': 'TCU0G2', 'label': 'TCU VER 0 GEN 2', 'group': 'E70'}, 'navigationAssistanceLevel': {'code': 'NAV3G5', 'label': 'LEVEL 3 TYPE 5 NAVIGATION', 'group': '408'}, 'battery': {'code': 'BT4AR1', 'label': 'BATTERIE BT4AR1', 'group': '968'}, 'radioType': {'code': 'RAD37A', 'label': 'RADIO 37A', 'group': '425'}, 'registrationCountry': {'code': 'FR'}, 'brand': {'label': 'RENAULT'}, 'model': {'code': 'X101VE', 'label': 'ZOE', 'group': '971'}, 'gearbox': {'code': 'BVEL', 'label': 'BOITE A VARIATEUR ELECTRIQUE', 'group': '427'}, 'version': {'code': 'INT MB 10R'}, 'energy': {'code': 'ELEC', 'label': 'ELECTRIQUE', 'group': '019'}, 'registrationNumber': 'ab12cd', 'vcd': 'SYTINC/SKTPOU/SAND41/FDIU1/SSESM/MAPSTD/SSCALL/SAND88/SAND90/SQKDRO/SDIFPA/FACBA2/PRLEX1/SSRCAR/CABDO2/TCU0G2/SWALBO/EVTEC1/STANDA/X10/B10/EA2/MB/ELEC/DG/TEMP/TR4X2/RV/ABS/CAREG/LAC/VT003/CPE/RET03/SPROJA/RALU16/CEAVRH/AIRBA1/SERIE/DRA/DRAP08/HARM02/ATAR/TEKNV/SFBANA/KM/DPRPN/AVREPL/SSDECA/ASRESP/RDAR01/ALEVA/CACBL2/SOP02C/CTHAB2/TRNOR/LVAVIP/LVAREL/SASURV/KTGREP/SGSCHA/APL03/ALOUCC/CMAR3P/NAV3G5/RAD37A/BVEL/AUTAUG/RNORM/ISOFIX/EQPEUR/HRGM01/SDPCLV/TLFRAN/SPRODI/SAN613/SSAPEX/GENEV1/ELC1/SANCML/PE2012/PHAS1/SAN913/045KWH/BT4AR1/VEC153/X101VE/NBT017/5AQ', 'assets': [{'assetType': 'PICTURE', 'renditions': [{'resolutionType': 'ONE_MYRENAULT_LARGE', 'url': 'https://3dv2.renault.com/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA2%2FDG%2FVT003%2FRET03%2FRALU16%2FDRAP08%2FHARM02%2FTEKNV%2FRDAR01%2FALEVA%2FSOP02C%2FTRNOR%2FLVAVIP%2FLVAREL%2FNAV3G5%2FRAD37A%2FSDPCLV%2FTLFRAN%2FGENEV1%2FSAN913%2FBT4AR1%2FNBT017&databaseId=1bc39d1c-ca60-4d83-9198-9bfc61d2da12&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_LARGE'}, {'resolutionType': 'ONE_MYRENAULT_SMALL', 'url': 'https://3dv2.renault.com/ImageFromBookmark?configuration=SKTPOU%2FPRLEX1%2FSTANDA%2FB10%2FEA2%2FDG%2FVT003%2FRET03%2FRALU16%2FDRAP08%2FHARM02%2FTEKNV%2FRDAR01%2FALEVA%2FSOP02C%2FTRNOR%2FLVAVIP%2FLVAREL%2FNAV3G5%2FRAD37A%2FSDPCLV%2FTLFRAN%2FGENEV1%2FSAN913%2FBT4AR1%2FNBT017&databaseId=1bc39d1c-ca60-4d83-9198-9bfc61d2da12&bookmarkSet=RSITE&bookmark=EXT_34_DESSUS&profile=HELIOS_OWNERSERVICES_SMALL_V2'}]}, {'assetType': 'PDF', 'assetRole': 'GUIDE', 'title': 'PDF Guide', 'description': '', 'renditions': [{'url': 'https://cdn.group.renault.com/ren/gb/myr/assets/x101ve/manual.pdf.asset.pdf/1558704861676.pdf'}]}, {'assetType': 'URL', 'assetRole': 'GUIDE', 'title': 'e-guide', 'description': '', 'renditions': [{'url': 'http://gb.e-guide.renault.com/eng/Zoe'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': '10 Fundamentals about getting the best out of your electric vehicle', 'description': '', 'renditions': [{'url': '39r6QEKcOM4'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Automatic Climate Control', 'description': '', 'renditions': [{'url': 'Va2FnZFo_GE'}]}, {'assetType': 'URL', 'assetRole': 'CAR', 'title': 'More videos', 'description': '', 'renditions': [{'url': 'https://www.youtube.com/watch?v=wfpCMkK1rKI'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Charging the battery', 'description': '', 'renditions': [{'url': 'RaEad8DjUJs'}]}, {'assetType': 'VIDEO', 'assetRole': 'CAR', 'title': 'Charging the battery at a station with a flap', 'description': '', 'renditions': [{'url': 'zJfd7fJWtr0'}]}], 'yearsOfMaintenance': 12, 'connectivityTechnology': 'RLINK1', 'easyConnectStore': False, 'electrical': True, 'rlinkStore': False, 'deliveryDate': '2017-02-07', 'retrievedFromDhs': False, 'engineEnergyType': 'ELEC'}}]}
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/12345678-3075-4da9-a711-8d49d69a6640/kamereon/kca/car-adapter/v2/cars/VFxxxxxx/battery-status?country=GB HTTP/1.1" 500 None
[... cut ...] ing', 'content-encoding': 'gzip', 'x-response-time': '129'}
--------------------  -------------
Battery level         Unavailable%
Range estimate        Unavailable
Plug state            Not available
Charging state        Not available
Charge mode           Unavailable
AC start at           Unavailable
External temperature  Unavailable
Total mileage         Unavailable
Updated at            Unavailable
--------------------  -------------
(homeassistant) homeassistant@webhome:/srv/homeassistant$

epenet, 2020-05-17:
And does it work correctly inside the Renault App?

epenet, 2020-05-17:
Also, once you get PyZe working, you might want to look at the [hassRenaultZE plugin for home assistant](https://github.com/epenet/hassRenaultZE) if you are planning to use it inside Home Assistant

OhSoGood, 2020-05-18:
epenet, thank you! You made me realized that my renault online services contract had expired. Pitifully they didn't send any email to warn me. Reactivation may take up to 72h, I'll keep you informed then.

jamesremuscat, 2020-06-01:
I'm going to close this, since it looks like pyze is behaving correctly for that situation. If you think it's not, please feel free to reopen this issue!

closed 2020-06-01 by jamesremuscat

epenet, 2020-06-01:
It would have been nice to hear from @OhSoGood to confirm his issues is fixed...

OhSoGood, 2020-06-01:
My apologies!
The fault was indeed with the contract not having been renewed and, after their renewal had been requested, there has been a failure in Renault's backoffice (contracts were officially running again, but technically API was not allowed for me... I had to escalate with Renault hotline).
The issue had never been in pyze. Now that's the API at Renault side is fine, pyze runs smoothly.