jamesremuscat/pyze #67: pyze via OpenHab

ZE135, 2020-05-22:
Hi all,

the script is working finde for me. The history is not working at all. Following expections I am receiving: 
`Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.4.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.4.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.4.0-py3.7.egg/pyze/cli/charge-history.py", line 20, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.4.0-py3.7.egg/pyze/cli/common.py", line 24, in get_vehicle
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.4.0-py3.7.egg/pyze/api/credentials.py", line 25, in inner
pyze.api.credentials.MissingCredentialException: kamereon-api-key
`

Does anybody integrated this script in openHab? this would be a big step forward. 

Thanks in advacne

ZE135, 2020-05-23:
I do not have good phyton skills. What I would like to do is to send the responses via REST to my OpenHAB server
Something like this 

`curl -X PUT --header "Content-Type: text/plain" --header "Accept: application/json" -d .format(status['batteryLevel']) "http://192.168.1.12:8080/rest/items/RenaultZEServices_Zoe_ChargeLevel/state"`

Any clue how I can integrate this?

ZE135, 2020-05-28:
I changed the header since it was missleading. 
The current response for the parameter charge-history is:

```
Error communicating with Renault API!
{"type":"FUNCTIONAL","messages":[{"code":"err.func.403","message":"{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access is denied for this resource\"}]}"}],"errors":[{"errorCode":"err.func.403","errorMessage":"{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access is denied for this resource\"}]}"}],"error_reference":"FUNCTIONAL"}
```
Does anybody has any idea? Status is working properly

ZE135, 2020-05-31:
Any clue? I guess the feature is not available anymore??? Does somebody has a clue if history is not supported anymore?

jamesremuscat, 2020-06-01:
I'm not sure what OpenHab is or what relevance it has to your report.

The error message indicates what might be wrong in your situation:

> pyze.api.credentials.MissingCredentialException: kamereon-api-key

Have you set `KAMEREON_API_KEY` as explained in the README?

> the script is working finde for me. The history is not working at all.

I don't know which "script" you're referring to. Without a Kamereon API key you won't be able to access any data about your vehicle.

Specifically with respect to the charge history endpoint: [it always returns an empty list for me, but others have had useful data returned](https://github.com/jamesremuscat/pyze/wiki/Known-issues-with-the-Renault-API#data-is-not-available).

ZE135, 2020-06-01:
Hi James, sorry, the header was missleading. I thought I changed it already last time. 
Changed it now. 

I used wrong terminology. To be more precize:

> pyze status --km 

Result:

 --------------------  ------------------------------------------------
Battery level         23%
Available energy      13kWh
Range estimate        87.0 km
Plug state            Unplugged
Charging state        Not charging
Charge rate           0.02kW
Time remaining        0:20
Charge mode           always
AC start at           Unavailable
External temperature  Unavailable
Battery temperature   20°C
Total mileage         Unavailable
Updated at            2020-05-31 13:58:55
Location              48.xxxxxx,8.xxxxxxx as of 2020-05-31 13:58:08
--------------------  ------------------------------------------------

Obviously the Kamereon_API_Key is correct, since partial results are fine (except ext. temp). 

Commands
> pyze ac-history
and
> pyze ac-stats

Results are identical:

> Error communicating with Renault API!
{"type":"TECHNICAL","messages":[{"code":"err.tech.501","message":"{\"errors\":[{\"status\":\"Not Implemented\",\"code\":\"501\",\"title\":\"Not supported Feature\",\"detail\":\"This feature is not technically supported by this gateway\"}]}"}],"errors":[{"errorCode":"err.tech.501","errorMessage":"{\"errors\":[{\"status\":\"Not Implemented\",\"code\":\"501\",\"title\":\"Not supported Feature\",\"detail\":\"This feature is not technically supported by this gateway\"}]}"}],"error_reference":"TECHNICAL"}

Commads 
> pyze charge-history
and
>  pyze charge-stats

Results are identical:

> Error communicating with Renault API!
{"type":"FUNCTIONAL","messages":[{"code":"err.func.403","message":"{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access is denied for this resource\"}]}"}],"errors":[{"errorCode":"err.func.403","errorMessage":"{\"errors\":[{\"status\":\"403\",\"code\":\"security.access\",\"detail\":\"Access is denied for this resource\"}]}"}],"error_reference":"FUNCTIONAL"}

Any idea why this happens? 

Note: I am not seing any charging history in the myRenault APP....

jamesremuscat, 2020-06-01:
The `501` error from `ac-stats` / `ac-history` is a problem at Renault's end; from memory, those endpoints have been broken in various ways for quite a while.

The `External temperature Unavailable` message might be related: someone at Renault thought it was a good idea to separate the functions of the API into two different subscriptions. In the UK those are now called "My Z.E. Connect" and "My Z.E. Interact", and you need both for some functions such as charge scheduling and remote preconditioning. Do you have active subscriptions to _both_ of those?

ZE135, 2020-06-01:
I am struggling to find how to find out if I have these subscriptions??
I the app I just can see in contracts:
"Connectivitiy Pack 36 month in base"
"Firmware update over the Air for life"
"Remote Car Dashboard 36 month in base"
"Pack Remote EV 36 month in base"

jamesremuscat, 2020-06-01:
> "Connectivitiy Pack 36 month in base"
> "Pack Remote EV 36 month in base"

It's hard to say for sure given Renault's confusing naming scheme and translating into English, but I _think_ those are the two you need.

In which case, it's a Renault problem if you've not got access to the data. Can you activate the car's air conditioning from the app? If not, get in touch with Renault support.

ZE135, 2020-06-01:
is there any chance to read out all available raw data? How can I see the available data from both subscriptions?

jamesremuscat, 2020-06-01:
Running pyze with the `--debug` argument will show you the raw data returned from the API.

There is no documentation as to which data is available from each type of subscription. Renault haven't published any documentation whatsoever about this API, and it's not something that's possible to figure out from the outside. Anything I say is merely (hopefully educated) guesswork.

ZE135, 2020-06-01:
Thanks James. 
I am seeing now lots of information in the debug-mode. 
Do you know e.g. which parameter is for external temp? or charging history? 

I would just crosscheck the data with the results

ZE135, 2020-06-01:
just to add a comment: I just reviewed the debug data visually. I could not find any charge_history data....

jamesremuscat, 2020-06-01:
I don't understand what you're expecting to see. We already know you're not going to see any `charge_history` data, because the Renault API returns an HTTP `403` when you try to access it.

ZE135, 2020-06-01:
Am I the only one who dont see charge_history data? This is confusing me. Is it a general problem or a specific one and I have to do something....?

jamesremuscat, 2020-06-01:
I don't work for Renault, so all I can tell you is that pyze is behaving correctly. Renault's server is not letting you access charge history data. I can't tell you why.

Those who _are_ able to access that endpoint have had mixed results (as per the link above).

ZE135, 2020-06-03:
Thanks anyway james. I opened a Renault ticket in order to get more information about.

Maybe one last question which is not really related to this issue, but I didnt want to open a new case: Is it normal that the "Time remaining" and "charge rate" are still visible, even if its unplugged and not charging?

BTW: What is the meaning of charge rate?

PysX, 2020-07-14:
Hello,
May be it's a ZE50 limitation. Same problem for me. You can look at #40 or #24 .

For the second point, Pyze returns what the Renault server transmits to him. Even when the car doesn't charge, Renault returns these values, which is why Pyze displays them. It's not a bug from Pyze, but from Renault.

You can have some information on charge rate here : #28

jamesremuscat, 2020-08-24:
Closing this since I don't think there are any outstanding actions.

closed 2020-08-24 by jamesremuscat