jamesremuscat/pyze #73: not working anymore

ik2wxx, 2020-07-01:
It's not working anymore, connection issues.
Worked until yesterday.
Maybe on july 1st they have changed something ?

```
Error communicating with Renault API!
{"type":"TECHNICAL","messages":[{"code":"err.tech.wired.kamereon-proxy","message":"Failed to forward request to remote service."}],"errors":[{"errorCode":"err.tech.wired.kamereon-proxy","errorMessage":"Failed to forward request to remote service."}],"error_reference":"TECHNICAL"}
```

jamesremuscat, 2020-07-01:
OK, so it looks like the "Kamereon token" is not only no longer required, but actively causes their servers to fall over if you include it in a request.

Nothing like making backwards-incompatible and broken changes to your live API that includes a version number to avoid making backwards-incompatible changes to it, eh? :facepalm: 

Fortunately that means the fix is easy... PR shortly.

closed 2020-07-01 by jamesremuscat

jamesremuscat, 2020-07-01:
@ik2wxx Thanks for bringing this to my attention - I've just released `v0.5.0` which includes the required fix.

ik2wxx, 2020-07-01:
Thank you !!!
Great support !!!

commit a05dfac0 (2020-07-01): Remove Kamereon token and don't include it in requests.

Renault have changed their API so that including a Kamereon token in a
request causes the server to error out. The solution, fortunately, is
straightforward; just remove all references to it from pyze.

Fixes #73.
PR #74 (2020-07-01): Remove Kamereon token and don't include it in requests.
Renault have changed their API so that including a Kamereon token in a request causes the server to error out. The solution, fortunately, is straightforward; just remove all references to it from pyze, since it no longer seems to be required.

Fixes #73.