jamesremuscat/pyze #77: Remove location from pyze status <ZE50

Dave-09, 2020-08-04:
If I run pyze status --km with my ZE40, I get 'Unavailable' on 'Location'. That's right!

So is it not possible, like with "Available energy" for example, to hide it from non ZE50?

Thank you!

jamesremuscat, 2020-08-04:
Slightly different mechanisms at play.

"Available energy" is part of an API call that also returns state of charge, estimated range, and a bunch of other things. The data returned by that API call changes, and we display whatever it contains.

"Location" is its own API call. If we call that API endpoint and it returns a value, we display it; if it does not, we display "Unavailable".

There's also nothing (obvious) in the API to say "I'm a ZE50" and certainly nothing to say "I'm a ZE50, use these API endpoints", so it's not as easy as saying "if car is not ZE50 do this".

Hope that clarifies things.

closed 2020-08-04 by jamesremuscat

Dave-09, 2020-08-05:
Understandably that isn't possible! :) otherwise it would have been a nice addition.

Thanks for your answer.