jamesremuscat/pyze #78: Unable to login

myhrmans, 2020-08-18:
As stated in [#43](https://github.com/epenet/hassRenaultZE/issues/43) we have started experiencing authentication error starting from today.
When trying to login using the correct credentials (tested by signing out and in of the android app) I get this error:
`raise RuntimeError( RuntimeError: Gigya returned error 403042: invalid loginID or password` 

Have Renault changed the authentication way?

g-mocken, 2020-08-18:
In pyze, it looks more like:
`RuntimeError: Gigya returned error 403007: Invalid namespace &#39;accounts&#39; or method &#39;getJWT&#39; or you do not have the required permissions to call it.`
In ZoeStatus, I get the same error for the method getAccountInfo instead of getJWT.
So far no idea what exactly is wrong ... the official MyRenault app is still working, though.

premultiply, 2020-08-18:
Similar problem here since today afternoon:

```
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.5.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/status.py", line 35, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/common.py", line 24, in get_vehicle
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/credentials.py", line 26, in inner
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/kamereon.py", line 119, in get_vehicles
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/credentials.py", line 26, in inner
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/gigya.py", line 109, in get_jwt_token
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/gigya.py", line 126, in raise_gigya_errors
RuntimeError: Gigya returned error 403007: Invalid namespace &#39;accounts&#39; or method &#39;getJWT&#39; or you do not have the required permissions to call it.
```

mpovel, 2020-08-18:
The same 403007 since about 14:00 MEST.

myhrmans, 2020-08-18:
@g-mocken @premultiply you are both signed in since before, correct? I get the same error on my previous signed in device.

MattCheetham, 2020-08-18:
Had about 20 reports through my app that uses the same API. Something has changed.

premultiply, 2020-08-18:
@MattCheetham The funny thing is that your Zeddy-App still works fine on my phone...

toby17780, 2020-08-19:
> Had about 20 reports through my app that uses the same API. Something has changed.

Yes, for me the Zeddy App also doesn't work anymore...

Raptoaaah, 2020-08-19:
Also broken on my devices - Maybe something changed with the introduction of the „new“ crappy MY Renault App, at least it occupies a fresh place in the App Store and can coexist with the „old“ new version

myhrmans, 2020-08-19:
> At least on my devices the „old“ My Renault App seems to have login issues too.

Old app? New App? Since when did they release a new app? I have the app "My Renault Sverige" from the playstore latest updated January 13 2020.

Dave-09, 2020-08-19:
There is indeed a 'new' My Renault app with a function "maintenance information". Everything else looks the same.

No idea why Renault had to install a whole new app if necessary. Incidentally, I immediately received a notification when opening the old app. You not @myhrmans?

Why is Renault continuously changing their API without any feedback information online? This is not getting anywhere.

epenet, 2020-08-19:
Looks like @jamesremuscat is looking at it: #79

premultiply, 2020-08-19:
Already fixed in EVCC https://github.com/andig/evcc/pull/301

McTristan, 2020-08-19:
would be great if somebody could compile a release as i have no access to git tools in the maschine I'm using it in (only pip and python3).

myhrmans, 2020-08-19:
I believe only @jamesremuscat can upload a new version to pipy, I created a local version to use for the moment.

myhrmans, 2020-08-19:
Fixed in #79

closed 2020-08-19 by myhrmans

jamesremuscat, 2020-08-19:
> would be great if somebody could compile a release as i have no access to git tools on the machine I'm using it in (only pip and python3).

I'll be compiling a release later today.

Dave-09, 2020-08-19:
Works fine now! 👍 

I just think Renault has put something back? because Zeddy (without update) and EVplugIN (without update) works again?

jamesremuscat, 2020-08-19:
Yes, looks like it. :man_facepalming: I guess they were just doing waiting until I made changes before undoing whatever it was they did! (I think it's a change on Gigya's end rather than Renault's).

Anyway, `v0.6.0` is now available on PyPI, and ought to continue to work if they change their minds again.

Dave-09, 2020-08-19:
Thank you again @jamesremuscat 👍

g-mocken, 2020-08-19:
I wonder why the behavior both  before and after the change have not much in common with the documented behavior here: https://developers.gigya.com/display/GD/accounts.getJWT+REST
Any idea why? Am I looking in the wrong place? Or simply expecting too much?
Anyway, thanks for sorting it out again, @jamesremuscat.