jamesremuscat/pyze #82: Replace RuntimeErrors in Gigya with appropiate (sub)classed Errors

mfonville, 2020-08-23:
At the moment, anything that goes wrong during the Gigya login process throws the same generic RuntimeError.
This makes it for users of the API impossible to detect if e.g. just login username/password supplied by the user is wrong, or if there is something else broken in e.g. a dependency.

Could we please change the error types, they could be a subclass of RuntimeError for more generic usage and backwards compatibility?
A good example of error usage is the Requests-library: https://requests.readthedocs.io/en/master/_modules/requests/exceptions/

My proposal would be a new exception class: `GigyaException(RuntimeError)`
Something similar could also be done for Kamereon. (or a PyZEException, but imho it would be appropriate to have them separate)

mfonville, 2020-08-28:
@jamesremuscat 
I would like to have some feedback how you feel about this, before I would make a PR :-)

epenet, 2020-10-27:
Hi @jamesremuscat 
Do you have a view on this?