jamesremuscat/pyze #83: Distinguish between various models

epenet, 2020-09-02:
Hi,

Do you think it is possible to use the "Kamereon.vehicles" response to work out what functionnality is available (or not available)?
In particular, it would be great to at least distinguish between ZE40 and ZE50.

This is what I have on ZE40:
```
			"vehicleDetails": {
				"vin": "XXXXX",
				"registrationDate": "YYYY-MM-DD",
				"firstRegistrationDate": "YYYY-MM-DD",
				"engineType": "5AQ",
				"engineRatio": "601",
				"deliveryCountry": {
					"code": "FR",
					"label": "FRANCE"
				},
				"family": {
					"code": "X10",
					"label": "FAMILLE X10",
					"group": "007"
				},
				"tcu": {
					"code": "TCU0G2",
					"label": "TCU VER 0 GEN 2",
					"group": "E70"
				},
				"navigationAssistanceLevel": {
					"code": "NAV3G5",
					"label": "LEVEL 3 TYPE 5 NAVIGATION",
					"group": "408"
				},
				"battery": {
					"code": "BT4AR1",
					"label": "BATTERIE BT4AR1",
					"group": "968"
				},
				"radioType": {
					"code": "RAD37A",
					"label": "RADIO 37A",
					"group": "425"
				},
				"registrationCountry": {
					"code": "FR"
				},
				"brand": {
					"label": "RENAULT"
				},
				"model": {
					"code": "X101VE",
					"label": "ZOE",
					"group": "971"
				},
				"gearbox": {
					"code": "BVEL",
					"label": "BOITE A VARIATEUR ELECTRIQUE",
					"group": "427"
				},
				"version": {
					"code": "INT MB 10R"
				},
				"energy": {
					"code": "ELEC",
					"label": "ELECTRIQUE",
					"group": "019"
				},
```

Could we use version (INT MB 10R for Zoe 40)?
Or maybe engineType/engineRatio (5AQ-601 for Zoe 40)?

What does it show for other users? I don't have other vehicles to compare...

jamesremuscat, 2020-09-02:
You'd hope that the information would be in that response somewhere!

Is yours a Q90 or an R90?

My ZE40 Q90 returns:

- Engine type: `5AM`
- Engine ratio: `450`
- Version code: `INT MA 17RP`
- Model code: `X101VE`

Would be good if we could build up enough data to work out which fields are actually useful here, and which car variants they correspond to.

fawick, 2020-09-02:
Happy to contribute! 

My ZE50 R135 returns:

- Engine type: `5AQ`
- Engine ration: `605`
- Version code: `INT A MC 1L`
- Model code: `X102VE`

kozmoz, 2020-09-02:
ZE40 R110 returns:

- Engine type: `5AQ`
- Engine ratio: `607`
- Version code: `LTD MF 18RP`
- Model code: `X101VE`

```javascript
      'vehicleDetails': {
        'vin': 'x',
        'registrationDate': '2020-06-16',
        'firstRegistrationDate': '2020-06-16',
        'engineType': '5AQ',
        'engineRatio': '607',
        'deliveryCountry': {
          'code': 'BE',
          'label': 'BELGIQUE'
        },
        'family': {
          'code': 'X10',
          'label': 'FAMILLE X10',
          'group': '007'
        },
        'tcu': {
          'code': 'TCU0G2',
          'label': 'TCU VER 0 GEN 2',
          'group': 'E70'
        },
        'navigationAssistanceLevel': {
          'code': 'NAV3G5',
          'label': 'LEVEL 3 TYPE 5 NAVIGATION',
          'group': '408'
        },
        'battery': {
          'code': 'BT4AR1',
          'label': 'BATTERIE BT4AR1',
          'group': '968'
        },
        'radioType': {
          'code': 'RAD06D',
          'label': 'RADIO 06D',
          'group': '425'
        },
        'registrationCountry': {
          'code': 'NL'
        },
        'brand': {
          'label': 'RENAULT'
        },
        'model': {
          'code': 'X101VE',
          'label': 'ZOE',
          'group': '971'
        },
        'gearbox': {
          'code': 'BVEL',
          'label': 'BOITE A VARIATEUR ELECTRIQUE',
          'group': '427'
        },
        'version': {
          'code': 'LTD MF 18RP'
        },
        'energy': {
          'code': 'ELEC',
          'label': 'ELECTRIQUE',
          'group': '019'
        },
        'yearsOfMaintenance': 12,
        'connectivityTechnology': 'RLINK1',
        'easyConnectStore': False,
        'electrical': True,
        'rlinkStore': False,
        'deliveryDate': '2018-09-19',
        'retrievedFromDhs': False,
        'engineEnergyType': 'ELEC',
        'radioCode': '0230'
      }
```

epenet, 2020-09-02:
It looks like we're on the right track with the model code.
- engine type seems to be always 5AQ
- engine ratio is probably to distinguish between R90 / R110 / R135
- version code add the flavours (LIFE / ZEN / INTENS / RIVIERA / LIMITED / ICONIC / etc.)

umbertones, 2020-09-02:
This is a Q210 from 2013 with the 41kWh battery upgrade
vehicleDetails': {
    'vin': 'y',
    'engineType': '5AM',
    'engineRatio': '450',
    'deliveryCountry': {
        'code': 'DE',
        'label': 'ALLEMAGNE'
        },
    'family': {
        'code': 'X10', 
        'label': 'FAMILLE X10', 
        'group': '007'
    }, 
    'tcu': {
        'code': 'TCU0G1', 
        'label': 'TCU VER 0 GEN 1', 
        'group': 'E70'
        }, 
        'navigationAssistanceLevel': {
            'code': 'NAV3G3', 
            'label': 'AIDE NAVIG 3 NIVEAU 3', 
            'group': '408'
        }, 
        'battery': {
        'code': 'BT4MR1', 
            'label': 'BATTERIE BT4MR1', 
            'group': '968'
        }, 
        'radioType': {
            'code': 'RAD37A', 
            'label': 'RADIO 37A', 
            'group': '425'
        }, 
        'registrationCountry': {
                    'code': 'DE'
        }, 
        'brand': {
            'label': 'RENAULT'
        }, 
        'model': {
            'code': 'X101VE', 
            'label': 'ZOE', 
            'group': '971'
        }, '
        gearbox': {
            'code': 'BVEL', 
            'label': 'BOITE A VARIATEUR ELECTRIQUE', 
            'group': '427'
        }, 
        'version': {
            'code': 'INT MA 7'
        }, 
        'energy': {
            'code': 'ELEC', 
            'label': 'ELECTRIQUE', 
            'group': '019'
        }
    }
The upgrade is not in the data. When you look in the vcd key that contains all the equipped stuff, you'll find:
026KWH/BT4MR1/VEC010/X101VE/NBT007/5AM

mfonville, 2020-09-02:
For my dashboard application I had collected some information:
Zoe Q210:
```
model: X101VE
engine code: 5AM
engine ratio: 450
battery: BT4MR1
```

Zoe R240:
```
model: X101VE
engine code: 5AQ
engine ratio: 601
battery: BT4MR1
```

Zoe ZE40 Q90:
```
model: X101VE
engine code: 5AM
engine ratio: 450
battery: BT4AR1
```

Zoe ZE40 R90:
```
model: X101VE
engine code: 5AQ
engine ratio: 601
battery: BT4AR1
```

Zoe ZE40 R110:
```
model: X101VE
engine code: 5AQ
engine ratio: 607
battery: BT4AR1
```

Zoe ZE50 R110:
```
model: X102VE
engine code: 5AQ
engine ratio: 607
battery: BT4AR1
```

Zoe ZE50 R135:
```
model: X102VE
engine code: 5AQ
engine ratio: 605
battery: BT4AR1
```

mfonville, 2020-09-02:
And for versions I have also collected some information, but the problem is that they differ sometimes per country. I have some notes about it, but they are in Dutch so not ready to share with a copy-paste

premultiply, 2020-09-02:
So decoding some of the values is easy.
X10 is Renaults car type code for ZOE. So X101xx is ZOE Ph1 ("ZE" and "ZE40") and X102xx is ZOE Ph2 ("ZE50").
5AQ is the Renault engine (22 kW charger), 5AM is the old and weak Continental engine with quick charger (43kW charger).

sunilson, 2020-09-04:
Austria/R135/ZE50/Zen

`"engineType":"5AQ","engineRatio":"605","deliveryCountry":{"code":"AT","label":"AUTRICHE"},"family":{"code":"X10","label":"BAUREIHE X10","group":"007"},"tcu":{"code":"AIVCT","label":"MIT AIVC-ZENTRALELEKTRIK","group":"E70"},"navigationAssistanceLevel":{"code":"SAN408","label":"CRITERE DE CONTEXTE","group":"408"},"battery":{"code":"BT4AR1","label":"BATTERIE BT4AR1","group":"968"},"radioType":{"code":"RA404","label":"A-IVIENDD, 2BO + 2BI + 2T, DOUBLE-MICRO, MIT SPCX, UKW1/DAB+UKW2","group":"425"},"registrationCountry":{"code":"AT"},"brand":{"label":"RENAULT"},"model":{"code":"X102VE","label":"ZOE","group":"971"},"gearbox":{"code":"BVEL","label":"ELEKTRISCHES VARIATOR-GETRIEBE","group":"427"},"version":{"code":"ZEN A MC 1L"},"energy":{"code":"ELEC","label":"ELEKTRISCH","group":"019"},`