jamesremuscat/pyze #84: Does charge mode work for some users?

epenet, 2020-09-02:
Hi everyone,

I can't seem to find a discussion regarding the charge_mode endpoint.
Currently it fails on my Zoe (ZE40 R90) : 500 Server Error: Internal Server Error on url...

Is this one of those endpoints that is only available on ZE50?

jamesremuscat, 2020-09-02:
Working for my ZE40 Q90:

```
DEBUG:urllib3.connectionpool:https://api-wired-prod-1-euw1.wrd-aws.com:443 "GET /commerce/v1/accounts/xxxxxxxxxxxxxx/kamereon/kca/car-adapter/v1/cars/xxxxxxxxxxxxxx/charge-mode?country=GB HTTP/1.1" 200 None
DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {"data":{"type":"Car","id":"xxxxxxxxxxxxxx","attributes":{"chargeMode":"schedule_mode"}}}
```

epenet, 2020-09-02:
Looks like my "ZE interactive" contract didn't renew properly...
My "Z.E. Connect" contract renewed ok, but ZE interactive is stuck in "activating"

I wonder if it's possible to get that information from PYZE. It appears in my contract list on the Renault app...

sunilson, 2020-09-04:
I think this isn't yet implemented in Pyze but I found the endpoint while building my own Zoe app:

https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/{accountId}/vehicles/{vin}/contracts?connectedServicesContracts=true&warranty=true&warrantyMaintenanceContracts=true&country=AT&lang=de&brand=RENAULT

Maybe something that can be implemented in the future in Pyze also :)