jamesremuscat/pyze #86: Enhancement idea - include Kameron response fields when starting pre-conditioning

TheBluProject, 2020-09-24:
Hi,
At the moment there's no return status when requesting AC start, but in the debug mode we can clearly see the Kameron response with some useful data that could be grabbed from there and included into the non-debug response from pyze to confirm that the request has been successful.

`DEBUG:pyze.api.kamereon:Received Kamereon vehicle response: {"data":{"type":"HvacStart","id":"03c48ea6-a923-46d3-89e2-19b432067a97","attributes":{"action":"start","targetTemperature":21.0}}}`

Even returning entire json as above would be better than no confirmation at all.

Thanks !

jumpjack, 2021-09-27:
Which is the url format? This (and others) is not working:

hvac-sessions?start=2021-09-26&end=2021-09-27

jamesremuscat, 2021-09-29:
> Which is the url format? This (and others) is not working:
> 
> hvac-sessions?start=2021-09-26&end=2021-09-27

Please could you stop asking questions that can be answered by [looking at the code](https://github.com/jamesremuscat/pyze/blob/develop/src/pyze/api/kamereon.py#L256) rather than posting on unrelated issues?

jumpjack, 2022-04-01:
> At the moment there's no return status when requesting AC start, but in the debug mode we can clearly see the Kameron response with some useful data

Server response does not says if car actually started HVAC, you need to check the value of **hvacStatus** in the output of endpoint **/hvac-status**:

```
{
    "id": "xxx MY VIN xxx",
    "socThreshold": 40,
    "lastUpdateTime": "2022-04-01T11:09:34Z",
    "hvacStatus": "off"  <<<<================== this value
}
```