jamesremuscat/pyze #87: precise documentation on "Obtaining API keys"

etobi, 2020-10-02:
The README.md says on "Obtaining API keys"

> You need two API keys: one for Kamereon and one for Gigya

When i tried to setup https://github.com/epenet/hassRenaultZE (which uses pyze) the keys i actually needed where `gigyaProd` and `wiredProd` (not `kamareon`).
It would be very helpful if the documentation would be more precise/explicit which keys to get.

jamesremuscat, 2020-10-03:
The documentation literally says:

> Look for:
> ```
> "wiredProd": {
>    "target": "https://api-wired-prod-1-euw1.wrd-aws.com",
> ```

I'm not sure how much more explicit I can be than that.

`kamereon` is the internal name of Renault's API.

closed 2020-10-03 by jamesremuscat

etobi, 2020-10-05:
Sure, the example shows `wiredProd`, but the JSON has also a key named `kamareon`, which isnt the one to use. This is what i stumbled about.

Why not helping new people not failing on the first try by saying e.g.

> You need two API keys: one for Kamereon (internal name of Renault's API, named `wiredProd` in the JSON) and one for Gigya (named `gigyaProd` in then JSON).