jamesremuscat/pyze #88: You have reached your quota limit

berti2911, 2020-10-22:
Good evening,
pyze works very fine for me but now i get the message "your have reached your quota limit".
Does anyone now what are the restrictions of accessing data?

I am calling the API every 30 seconds to get the battery level for dynamical loading while checking the photolatic overflow. So i call it 2.880 times a day. This works for many month but now i get the above message.

edddeduck, 2020-10-22:
I think they just noticed you’ve been abusing the system and have put a block on it. 

Given the slow speed of charging using photovoltaic running the battery status every 30 minutes would likely suffice and keep under the rate limits. I doubt the car will even update any useful difference within that time period.

premultiply, 2020-10-22:
The update interval from the car during charge is only about 30 minutes. It is absolutly useless to poll the API that frequently.

To use well done PV-charging with proper car API implementation and Soc-interpolation use a good open source charge controller like https://github.com/andig/evcc

berti2911, 2020-10-22:
Thanks for your quick answers, i will now reduce the frequence of polling and call the API only for 2 hours in the morning.

As you can see in my log-File I check every morning at 6 am if there is enough power to drive to my office. Actually i check every 10 minute, which worked very fine.

2020-10-16 06:00:35,384 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 2.0 kWh
2020-10-16 06:10:51,082 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 2.0 kWh
2020-10-16 06:21:02,099 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 4.4 kWh
2020-10-16 06:31:12,338 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 4.4 kWh
2020-10-16 06:41:22,831 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 6.8 kWh
2020-10-16 06:51:33,939 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 9.2 kWh
2020-10-16 07:01:43,923 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 9.2 kWh
2020-10-16 07:11:53,163 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 12.0 kWh
2020-10-16 07:22:03,274 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 14.4 kWh
2020-10-16 07:32:13,766 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 14.4 kWh
2020-10-16 07:42:26,619 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 17.2 kWh
2020-10-16 07:52:37,725 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 19.6 kWh
2020-10-16 08:02:47,837 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 19.6 kWh
2020-10-16 08:12:57,175 - root - DEBUG -                Result Funktion kfz_get_ladung, KfZNr 1 Ladung: 22.0 kWh

berti2911, 2020-10-22:
Also i looked at the EV Charge Controller and saw that it supports my go_echarger. But I can not imagine how to get the battery level with this API because the wallbox can't read the battery level of the car.

premultiply, 2020-10-22:
Just try it out an see how it works.
It connects the data from metering, inverters, home batteries, chargers and vehicles together.
It does also support soc charge limits, minimum soc charging, and target charging (in test).

kakish, 2020-10-24:
> The update interval from the car during charge is only about 30 minutes. It is absolutly useless to poll the API that frequently.
> 
> To use well done PV-charging with proper car API implementation and Soc-interpolation use a good open source charge controller like https://github.com/andig/evcc

You are not right, update interval for ZOE ZE50 is approx every minute, so you can quite close monitor charging process. Or rather you could as of restrictions made by Renault.

mountbatt, 2020-10-25:
I have also come within the quota limit. How long do I have to wait approximately until it works again? Do you have any experience with this?

berti2911, 2020-10-26:
Sorry, no experience with this. I tried first to make seldom requests but still not works. Now i stoppt it completely and had to wait some time.
Does anyone know how to contact Renault with this topic?

berti2911, 2020-10-26:
Now it is working, i've waited for 4 hours without any API request.