jamesremuscat/pyze #90: Extra features for the PyZE package

epenet, 2020-10-31:
Hi @jamesremuscat,

I am wondering if you would be willing to opening up your repository to more collaborators, or moving your repository to a GitHub organisation.

I am guessing that the current implementation of PyZE fits your needs, and you do not feel any need to add extra functionnality to the library, so it has gone onto the back burner on your side.

If we ignore PR #70 (which is quite complex), I have a couple of small PRs (#80 and #85) which have been opened for a couple of months, and haven't received any attention.
On issue #82, the user is offering to add some features, and just needs a green light but wants confirmation that you agree on principle.
There are also some feature requests (#32, #43, #54, #83, #84, #86) that do not seem to have anyone working on, but I think it's only because other users are not sure if corresponding PR would be merged in.

As for my personal needs: in order to add the "renault" integration into Home Assistant core, I need to move some of the features from my hassRenaultZE custom component into a package hosted on pypi.org, and I need to device if I:
1. provide more PRs onto your repository, which may never get merged
2. create a new intermediate package which depends on your package and adds extra functionnality
3. create a new independant package which replicates all the API features from your package and adds extra functionnality

Please tell me what your plan is with the PyZE package, and hopefully we can work together to improve it some more.
I also know of other developers who would be interested to help move it forward.

Regards,
Erwann PENET

mfonville, 2020-11-01:
Thanks for opening this question @epenet, I was wondering the same.
Because PyZE is a good library, and it would be best to keep development coherent and keeping focus on one well-developed library for all. And keeping the documentation + experimental data at central place.

I am don't have the time available to put major development into PyZE, and indeed with my issue I wanted to consult first before investing the time into a PR. But I do have an interest to at least be involved with keeping the current codebase alive and add some minor development.

mfonville, 2020-11-24:
@epenet I can't find any of your contact details via GitHub, would you mind sending me an e-mail to get in direct contact? Thanks!

oncleben31, 2020-11-24:
I think you can reach him on Discord and in the [HACF network](https://github.com/hacf-fr/Information) (mainly french speaking)

oncleben31, 2020-11-24:
And there is a new package in development https://github.com/hacf-fr/renault-api

epenet, 2020-11-25:
Thanks @oncleben31 
@mfonville if you really wish to use email, you can use dev_at_zeflip_dot_com