jamesremuscat/pyze #91: PyZE doesn't work with latest Zoe Easylink software anymore

mfonville, 2020-11-26:
This week both my Zoe ZE50 and the MyRenault app of my phone received an update.
PyZE  doesn't get any more up-to-date telemetry information, but MyRenault app works fine :-/.

How can we reverse engineer any changes that Renault has made? @jamesremuscat

edddeduck, 2020-11-26:
I have an automated system running and it's updating as expected & I can see my driving odometer update from this morning and the battery drop.

UK Based cars testing on one ZE50 and one ZE 40 vehicle.

mfonville, 2020-11-26:
Hmm, so weird.
Also it seemed for me to go phased. First PyZE still reported some values, if I opened the MyRenault app to update, it would also make that value available to PyZE. But at the moment it doesn't get any value anymore.

My car is The Netherlands.

@edddeduck did your ZE50 already got the "283C30559R" system update that is available since the start of this week?

mfonville, 2020-11-26:
Hmm, another single telemetry point came through again at 14:17 CET (after I had opened the MyRenault app on my phone)
It is really weird...

mfonville, 2020-11-26:
I am getting some more telemetry datapoints now even without using the app.

Still it is weird though (and can't find an explanation at this moment) why throughout the day I could get updates on the app, while the PyZE scripts did not get those. :-/

mfonville, 2020-11-28:
Ok, I decompiled the latest MyRenault app and don't see any changes in the API.
So can't explain really why this happened, maybe AWS problems or so. I will close it and just hope it doesn't happen again...

closed 2020-11-28 by mfonville