jamesremuscat/pyze #92: Twingo ZE - KeyError: 'chargeMode'

TheReiner, 2021-01-16:
Hi, it work very well with my Zoe but i buy a new car... An Electric Twingo ZE and i have this error 

Thanks

/Renault# pyze status --vin VF1AXXXXXXXXXXXXXX
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.5.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/status.py", line 63, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/kamereon.py", line 202, in charge_mode
KeyError: 'chargeMode'

TheReiner, 2021-01-16:
Same problem with 0.60

Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.6.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/cli/status.py", line 63, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/api/kamereon.py", line 202, in charge_mode
KeyError: 'chargeMode'

TheReiner, 2021-01-16:
I find a solution....
If you see on My renault APP, charge management is on Unknow

[image]

So i think for the case to add Unknow... 
And if you have this error just put this command 

pyze charge-mode --always --vin VF1XXXXXXXXXXXX

epenet, 2021-02-03:
@TheReiner is this still occuring?
Did you get a chance to get a trace of the full json?

TheReiner, 2021-02-03:
Hi Epenet,
i don't know why... but since 3 days, i haven't "Unknow" now... 
Maybe wait for another case :)

epenet, 2021-02-03:
If it happens again it would be great to have a trace, so that I can adjust [renault-api](https://github.com/hacf-fr/renault-api) accordingly..

TheReiner, 2021-02-03:
I thinkI think I had this error because my car was new, and I think it was partially active...
So I think I will no longer have the error and that we have to wait for a new buyer :)

berti2911, 2022-08-08:
Here is a new buyer :-)
I had the same problem and with the hint of your post i know now that the unknown charge mode message disappears when you select the charge mode **one time** in your car (settings/EV programming/charge mode).
Same for climate contol