jamesremuscat/pyze #94: Error communicating with Renault API!

plin2, 2021-02-01:
Hi,

I'm getting an error message when I start pyze:

Error communicating with Renault API!
{"error":"access_denied","error_description":"Unauthorized"}

Did Renault change the API or did the API keys get invalidated`?

db-EV, 2021-02-01:
See [Error 403 on person](https://github.com/hacf-fr/renault-api/issues/181)

plin2, 2021-02-02:
New KAMERON API Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2 works for me. Where do you find it? 

I tried the keys obtained through 
https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_en_GB.json
but they didn't work.

db-EV, 2021-02-02:
Se [https://gist.github.com/mountbatt/772e4512089802a2aa2622058dd1ded7#gistcomment-3615860](https://gist.github.com/mountbatt/772e4512089802a2aa2622058dd1ded7#gistcomment-3615860) and below.

plin2, 2021-02-02:
new key works, will follow the other thread

jumpjack, 2022-03-16:
Key changed again with recent app update? (21/2/2022)

plin2, 2022-03-16:
I have access problems as well:
{"error":"access_denied","error_description":"Unauthorized"}

jumpjack, 2022-03-16:
I just replicated the issue also in my own app and in https://github.com/hacf-fr/renault-api , hence it's not a pyze bug.

plin2, 2022-03-16:
I tried to get fresh keys from https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_en_DE.json and extracted the "wiredProd" and "gigyaProd" apikeys - didn't help. My RenaultZE App seams to work.

db-EV, 2022-03-16:
I think it's a new Kamereon API key we need. This one is hardcoded in the app if I remember correct.

plin2, 2022-03-16:
The new KAMEREON API Key is **VAX7XYKGfa92yMvXculCkEFyfZbuM7Ss**

closed 2022-03-16 by plin2

db-EV, 2024-03-05:
No, ZoePHP and HA still working.