jamesremuscat/pyze #95: Error communicating with Renault API

timmchugh11, 2021-02-03:
Hello, 

About 2 days ago my connection stopped working and I get the error,

Error communicating with Renault API!
{"error":"access_denied","error_description":"Unauthorized"}

Is this an issue with pyze? I have tried re adding my credentials and I get the same error when logging in.

Thanks

epenet, 2021-02-03:
There was a change of Key on the Renault servers.
There is a discussion about it on https://github.com/hacf-fr/renault-api/issues/181, with corresponding pull request https://github.com/hacf-fr/renault-api/pull/182

timmchugh11, 2021-02-03:
Thank you, I just used the new key and all is working again

The key I used from the link you sent, Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2

yannickh, 2021-02-08:
Hmm, I got it working again with the new key, but it seems to have broken again?

premultiply, 2021-02-08:
Correct. Something has changed again but https://github.com/hacf-fr/renault-api still works.
So let us find the difference.

db-EV, 2021-02-08:
Problem maybe related to https://github.com/jamesremuscat/pyze/issues/78?

premultiply, 2021-02-08:
What exactly to you mean?

epenet, 2021-02-08:
Note: https://github.com/hacf-fr/renault-api still works, but in my case the data hasn't changed since 2021-02-07T12:17:58+01:00 and no longer reflects the reality.

premultiply, 2021-02-08:
Ok, I found the problem in EVCC code. Was not a problem on Renaults side...
So API (with updated Kamereon key) still works fine here. After vehicle wakeup the data was updated correctly.

db-EV, 2021-02-08:
The latest version of pyze should work I think. But there is an unnecessary step at login (calling accounts.getAccountInfo), maybe this is causing problems?
But on a German forum pyze is working but the Zeddy app not.

db-EV, 2021-02-08:
> 
> 
> Ok, I found the problem in EVCC code. Was not a problem on Renaults side...
> So API (with updated Kamereon key) still works fine here. After vehicle wakeup the data was updated correctly.

Yes, it seems to be some wrong Gigya integrations out there.

premultiply, 2021-02-08:
Not true. Just the commit with new key has been accidentally overwritten.

fungus75, 2021-02-08:
> > Ok, I found the problem in EVCC code. Was not a problem on Renaults side...
> > So API (with updated Kamereon key) still works fine here. After vehicle wakeup the data was updated correctly.
> 
> Yes, it seems to be some faulty Gigya integrations out there.

Starting with today 12:00 CET gigya changed (or never accepted) some parameter-names. See (sorry german) https://github.com/fungus75/ioBroker.zoe2/issues/17 or have a look at the implementation of the gigya-part here: https://github.com/hacf-fr/renault-api/issues

fungus75, 2021-02-08:
Please find here which parameter I had to adjust: https://github.com/fungus75/ioBroker.zoe2/commit/c787fda42de51d7f7df4a087706ed1a67e55cd9c

premultiply, 2021-02-08:
Well... but that is not the case for the very clean EVCC implementation here: https://github.com/andig/evcc/blob/master/vehicle/renault.go

db-EV, 2021-02-08:
And calling account.getAccount is unnecessary, it's enough with accounts.login and accounts.getJWT.
You get the needed gigyaPersonID already on login.

db-EV, 2021-02-08:
And it was not the case for pyze either, there are other scripts, apps and so on with faulty Gigya integration.

kozmoz, 2021-02-09:
So what's the fix to have `pyze` working again? Obviously some people digged into the problem.

yannickh, 2021-02-09:
In the Gigya calls (getAccountInfo and getJWT), the parameter "oauth_token" should be changed to "login_token".

epenet, 2021-02-09:
> So what's the fix to have `pyze` working again? Obviously some people digged into the problem.
@kozmoz you don't describe your problem so it might not be linked to this discussion

> In the Gigya calls (getAccountInfo and getJWT), the parameter "oauth_token" should be changed to "login_token".
@yannickh it is already the case in pyze since #79 was merged last year

kozmoz, 2021-02-09:
> > So what's the fix to have `pyze` working again? Obviously some people digged into the problem.
> 
> @kozmoz you don't describe your problem so it might not be linked to this discussion
> 

Exactly the same error as @timmchugh11 who started this issue.

`Error communicating with Renault API!`
`{"error":"access_denied","error_description":"Unauthorized"}`

premultiply, 2021-02-09:
You need to update the Kamereon API key!!!

kozmoz, 2021-02-09:
Alright. Missed the new key for Kamereon, but did updated the Gigya API key. 
In summary, this fixed my issue:

```
export KAMEREON_API_KEY="Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2"
export GIGYA_API_KEY="3_e8d4g4SE_Fo8ahyHwwP7ohLGZ79HKNN2T8NjQqoNnk6Epj6ilyYwKdHUyCw3wuxz"
```

brusch, 2021-02-11:
I'm still facing issues: 
```
Traceback (most recent call last):
  File "/opt/bin/pyze", line 8, in <module>
    sys.exit(main())
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/cli/__main__.py", line 69, in main
    parsed_args.func(parsed_args)
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/cli/status.py", line 35, in run
    v = get_vehicle(parsed_args)
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/cli/common.py", line 24, in get_vehicle
    vehicles = k.get_vehicles().get('vehicleLinks')
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/api/credentials.py", line 26, in inner
    return func(*args, **kwargs)
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/api/kamereon.py", line 114, in get_vehicles
    self.get_account_id(),
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/api/kamereon.py", line 73, in get_account_id
    accounts = self.get_accounts()
  File "/volume1/Share/Web/SmartHome/.local/lib/python3.8/site-packages/pyze/api/credentials.py", line 25, in inner
    raise MissingCredentialException(name)
pyze.api.credentials.MissingCredentialException: gigya-person-id
```

any ideas?

Maje-Ax, 2021-02-11:
Issue was fixed with provided new credentials.
But today, we are facing a new authent issue with Gigya:

Traceback (most recent call last):
  File "myrenault.py", line 45, in <module>
    g.account_info()  # Retrieves and caches person ID
  File "/opt/myrenault/pyze/api/credentials.py", line 26, in inner
    return func(*args, **kwargs)
  File "/opt/myrenault/pyze/api/gigya.py", line 77, in account_info
    raise_gigya_errors(response_body)
  File "/opt/myrenault/pyze/api/gigya.py", line 126, in raise_gigya_errors
    response_body.get('errorDetails')
RuntimeError: Gigya returned error 403007: Invalid namespace &#39;accounts&#39; or method &#39;getAccountInfo&#39; or you do not have the required permissions to call it. 

Same error if using previous gigya API or new one.

Any ideas ?

Podariss, 2021-02-12:
Same here;
<code>
pi@TestPi:~/pyze$ pyze status --km
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.5.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/cli/status.py", line 35, in run
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/cli/common.py", line 24, in get_vehicle
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/api/credentials.py", line 26, in inner
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/api/kamereon.py", line 119, in get_vehicles
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/api/credentials.py", line 26, in inner
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/api/gigya.py", line 109, in get_jwt_token
  File "/usr/local/lib/python3.5/dist-packages/PyZE-0.5.0-py3.5.egg/pyze/api/gigya.py", line 126, in raise_gigya_errors
RuntimeError: Gigya returned error 403007: Invalid namespace &#39;accounts&#39; or method &#39;getJWT&#39; or you do not have the required permissions to call it.
</code>

Used the mentioned api keys from above and from keys mentioned in the readme files

db-EV, 2021-02-12:
Are you using the correct Gigya API keys for your country? These are different for each country.
Only the Kamereon API key is static for all countries.

Podariss, 2021-02-12:
I used the GB version before, but even with nl_NL no change...

berti2911, 2021-02-12:
Hi, i had the same issue and now it workes with the keys of kozmos.
Then i played a bit with the keys. Here are my results: the Gigya-key is not so strikt, for me also the english and the german version worked:
English

		"gigyaProd": {
			"target": "https://accounts.eu1.gigya.com",
			"apikey": "3_e8d4g4SE_Fo8ahyHwwP7ohLGZ79HKNN2T8NjQqoNnk6Epj6ilyYwKdHUyCw3wuxz"

German

        "gigyaProd": {
            "target": "https://accounts.eu1.gigya.com",
            "apikey": "3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668"

but the Kamereon Key is bitchy: neither the german nor the english version was successfull. Here i got the error message:

"Error communicating with Renault API!
{"error":"access_denied","error_description":"Unauthorized"}"

Only the Kammereon key of kozmos worked. So a question to him: how did you get this key?

epenet, 2021-02-12:
Key retrieval has changed on the Renault side.
It is now hidden deep inside Firebase, and for now you need to use Fiddler to extract it.

There is a discussion on https://github.com/hacf-fr/renault-api/pull/184

berti2911, 2021-02-12:
@Podariss, @brusch: maybe it helps to delete first the lokal file .credentials/pyze.json before starting again with pyze login?

kozmoz, 2021-02-12:
> Only the Kammereon key of kozmos worked. So a question to him: how did you get this key?

In the third message of this thread. Initially I also missed that one, but later I realised that was the fix for access_denied.

https://github.com/jamesremuscat/pyze/issues/95#issuecomment-772813562

heckmic, 2021-02-22:
Same Problem here:

`Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.5.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/status.py", line 35, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/cli/common.py", line 24, in get_vehicle
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.5.0-py3.7.egg/pyze/api/credentials.py", line 25, in inner
pyze.api.credentials.MissingCredentialException: kamereon-api-key
`

berti2911, 2021-02-22:
@Heckmic: Did you use exactly the following KEY? Only this one worked for me.
export KAMEREON_API_KEY="Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2"

And of course you have to login again: pyze login

jumpjack, 2021-09-21:
I am using the updated key  Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2 , but I am still getting this error:

{"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unauthorized","message":"The access is unauthorized"}],"errors":[{"errorCode":"err.func.wired.unauthorized","errorMessage":"The access is unauthorized"}],"error_reference":"FUNCTIONAL"}

I am using both direct url call and GET/POST using a Rest API Chrome extension, no luck.

My direct link:
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/persons/MY_PERSON_ID?apikey=Ae9FDWugRxZQAGm3Sxgk7uJn6Q4CGEA2&x-gigya-id_token=MY_JWT_TOKEN&country=IT

jumpjack, 2021-09-21:
I also know that there is a direct link which gives a very long JSON response, containing a lot of data, including my VIN.... but I found it by chance after dozens of attempts and I cannot get it anymore, any idea how it could be formed?

berti2911, 2021-09-21:
Are you using pyze? If not maybe you could download this library and look in the code how the web-service or link is formed.
For pyze it is essential to delete first the file .credentials/pyze.json and then login again with "pyze login" to avoid old chaching.

fungus75, 2021-09-21:
Have a look at that: https://github.com/fungus75/ioBroker.zoe2/blob/master/main.js 
Rewritten pyze (and some more projects) in Javascript for usage within iobroker.
Maybe it helps?

jumpjack, 2021-09-21:
This is what I discovered till now:

Download api keys GIGYA_KEY and KAMEREON_KEY:

https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_it_IT.json

Download Gigya session cookie (cookieValue):

https://accounts.eu1.gigya.com/accounts.login?loginID=MAIL&password=PASSWORD&apikey=GIGYA_KEY

Download Kamereon Person Id (PersonId):

https://accounts.eu1.gigya.com/accounts.getAccountInfo?apikey=GIGYA_KEY&login_token=COOKIEVALUE

Download Gigya JWT token (Id_token):

https://accounts.eu1.gigya.com/accounts.getJWT?apikey=GIGYA_KEY&login_token=COOKIEVALUE

Then query Kamereon server in some way which I don't understand yet...

To get the account(s) associated to a personId:

https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/persons/PERSONID?country=IT

To get vehicles associated to an account:

https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/ACCOUNTID/kamereon/kca/car-adapter/v1/cars/VEHICLE_VIN/location?country=IT

renovelles, 2022-03-18:
Dear all, Since yesterday, I have exactly the same issue than timmchugh11 last year. Someone to help me?

AllanPetersenDK, 2023-04-22:
What is LoginID and password for GIGYA?
Is it login for my Renault app?

jumpjack, 2023-04-25:
I think the APi key has just changed again, breaking all the custom apps.

jumpjack, 2023-05-23:
This should be the new kamereon api key:

YjkKtHmGfaceeuExUDKGxrLZGGvtVS0J

Source: https://gist.github.com/mountbatt/772e4512089802a2aa2622058dd1ded7