jamesremuscat/pyze #98: 500 Server Error: Internal Server Error

dvanders, 2021-03-25:
Since around 30 minutes ago I'm getting a 500 error on the battery status.

```
requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxx/kamereon/kca/car-adapter/v2/cars/yyy/battery-status?country=GB
```

And the cli is all unavailable
```
# pyze status --km --kw
--------------------  -------------
Battery level         Unavailable%
Range estimate        Unavailable
Plug state            Not available
Charging state        Not available
Charge mode           Unavailable
AC start at           Unavailable
External temperature  Unavailable
Total mileage         Unavailable
Updated at            Unavailable
Location              Unavailable
--------------------  -------------
```

Just me?

ledermann, 2021-04-16:
Same here, but only since today. The Renault app on iOS fails, too.

premultiply, 2021-04-16:
The application did run too long on Renaults cloud servers.
They have to put in some more coins to start it again...

ledermann, 2021-04-17:
For me, it works again since today.

jvrietveld, 2021-05-16:
No service here since 30 April.

epenet, 2021-05-17:
If you add debug option to the CLI, maybe you will get the underlying reason.
If the issue lasts for more than a day it's probably an issue with your account.

jvrietveld, 2021-05-17:
The debug option revealed it: "You have reached your quota limit".  It looks like my car is getting old (2016) and I have to renew a subscription. Last year I could renew this for a year but now I cannot find this option anymore.