jamesremuscat/pyze #109: Pyze is not working anymore (16-MAR-22?)

staudtm, 2022-03-20:
I heard that Renault has changed the API (or introduced a new API key?). Can someone please check?

staudtm, 2022-03-20:
I found this hint in a german forum (OpenWB):

neuem api-key.

$kamereon_api = 'VAX7XYKGfa92yMvXculCkEFyfZbuM7Ss';

RalfJahns, 2022-03-20:
Thank you very much to staudtm! Was a quick fix because of this post.

staudtm, 2022-03-20:
Hi Ralf, can you tell how you fixed it in detail? Do we have to change the source code?

RalfJahns, 2022-03-20:
No, you just have to use the new key.

From "CLI quickstart" section:

export KAMEREON_API_KEY=oF09WnKqvBDcr...
export GIGYA_API_KEY=3_e8d4g4SE_Fo8ahyH...

pyze login  # You should only need to do this once

You just have to do this (again).

TheReiner, 2022-03-24:
Same problem for me.. Until 16-MAR-22
But this solution doesn't work for me :-(

TheReiner, 2022-03-26:
> 

Work fine with this Key ;)

TheReiner, 2022-03-26:
> I found this hint in a german forum (OpenWB):
> 
> New api-key.
> 
> $kamereon_api = 'VAX7XYKGfa92yMvXculCkEFyfZbuM7Ss';

Thanks, Works fine with this new key :)

berti2911, 2022-08-07:
Thanks to "TheReiner" for presenting this new kamereon key. I searched again at least 3 hours for solving the authorisation  problem. The new key works but who long.. this is a real shit.

saxos1983, 2022-08-22:
Thank you very much TheReiner for finding this out.