jamesremuscat/pyze #111: No matching distribution found for renault-api

ako-3004, 2022-09-18:
Hello :) I use a new installed raspberry. did the following:

```
python3 --version
Python 3.7.3

python3 -m pip --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)

python3 -m pip install --upgrade pip setuptools wheel

pip install renault-api

```
resulting in

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting renault-api
  Could not find a version that satisfies the requirement renault-api (from versions: )
No matching distribution found for renault-api

```
Am I doing something wrong?

jamesremuscat, 2022-09-19:
I don't know what the `renault-api` package is that you're trying to install, but it has nothing to do with me. This GitHub repo is for `pyze`.

closed 2022-09-19 by jamesremuscat

ako-3004, 2022-09-19:
Sorry for that, I tried different renault apis and just mixed up the open browser tabs...
This one was for https://github.com/hacf-fr/renault-api/issues :D