jamesremuscat/pyze #113: How to obtain API keys? (I thought I understood ... but seems I do something wrong)

RalfJahns, 2023-01-23:
Hello jamesmuscat (or anybody else who wants to help),

thank you so much for your reply to issue #112

I followed the steps in "Obtaining API keys"-section in the Readme.md.
I cleared my browsers (firefox) cache and opened the URL https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_en_GB.json. I get this:

	"servers": {
		"wiredProd": {
			"target": "https://api-wired-prod-1-euw1.wrd-aws.com",
			"apikey": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB"
		},
		"gigyaProd": {
			"target": "https://accounts.eu1.gigya.com",
			"apikey": "3_e8d4g4SE_Fo8ahyHwwP7ohLGZ79HKNN2T8NjQqoNnk6Epj6ilyYwKdHUyCw3wuxz"
		}
	},

The wiredProd-key (this should be the kameron key, right?) is the one, **that you also mentioned in the Readme.md but not the one that you mentioned in** #112: VAX7...7Ss.

Earlier, I always tried the URL with de_DE at the end. Then I get:
        "wiredProd": {
            "apikey": "oF09WnKqvBDcrQzcW1rJNpjIuy7KdGaB"
        "gigyaProd": {
            "apikey": "3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668"
These keys are not working. Tried this 50 times last week!

Any help would be highly appreciated
Ralf Jahns

epenet, 2023-01-23:
Those keys are outdated

RalfJahns, 2023-01-23:
> Those keys are outdated

And that's the reason why I tried to get the currently valid ones (as I wrote) by following the steps in the Readme.md of this project.

epenet, 2023-01-23:
See https://github.com/hacf-fr/renault-api/blob/ad76a1bc463f3f6eeb11e056eb0ae4df3e17a1e1/src/renault_api/const.py#L21

epenet, 2023-01-23:
In particular https://github.com/hacf-fr/renault-api/pull/552

RalfJahns, 2023-01-23:
O.K., the new keys work. Thank you very much!
I feel a little dpendent, now that I've found out that I can't update the keys by myself :-(

epenet, 2023-01-23:
Blame it on Renault. They try to hide the keys now