jamesremuscat/pyze #115: Error access_denied / Unauthorized since this morning 18/04/2023

TheReiner, 2023-04-18:
i try to change the Key... but same problem.
{"error":"access_denied","error_description":"Unauthorized"}

i'am alone ?

TheReiner, 2023-04-18:
i try with this API KEY

export KAMEREON_API_KEY=Z7P4xyNrTITzNZh52oObSOVfKns9XGLE

`Enter your password: 
Error communicating with Renault API!
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 169, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 73, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 382, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 1012, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 353, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7fe13573ddc0>: Failed to establish a new connection: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.eu1.gigya.com', port=443): Max retries exceeded with url: /accounts.login (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe13573ddc0>: Failed to establish a new connection: [Errno -2] Name or service not known'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/PyZE-0.6.0-py3.9.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.9/dist-packages/PyZE-0.6.0-py3.9.egg/pyze/cli/login.py", line 14, in run
  File "/usr/local/lib/python3.9/dist-packages/PyZE-0.6.0-py3.9.egg/pyze/api/gigya.py", line 34, in login
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 590, in post
    return self.request('POST', url, data=data, json=json, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='accounts.eu1.gigya.com', port=443): Max retries exceeded with url: /accounts.login (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe13573ddc0>: Failed to establish a new connection: [Errno -2] Name or service not known'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 33, in <module>
    sys.exit(load_entry_point('PyZE==0.6.0', 'console_scripts', 'pyze')())
  File "/usr/local/lib/python3.9/dist-packages/PyZE-0.6.0-py3.9.egg/pyze/cli/__main__.py", line 72, in main
AttributeError: 'NoneType' object has no attribute 'text'`

TheReiner, 2023-04-18:
The new key is YjkKtHmGfaceeuExUDKGxrLZGGvtVS0J

yannick-ha, 2023-04-20:
Nice, thanks

ZE135, 2023-07-06:
Having similar issue. 
I used following keys:
export GIGYA_API_KEY=3_oRlKr5PCVL_sPWUZdJ8c5NOl5Ej8nIZw7VKG7S9Rg36UkDszFzfHfxCaUAUU5or2
export KAMEREON_API_KEY=YjkKtHmGfaceeuExUDKGxrLZGGvtVS0J
Did a login (which was successfull) and received following error message: 
Traceback (most recent call last):
  File "/usr/local/bin/pyze", line 11, in <module>
    load_entry_point('PyZE==0.6.0', 'console_scripts', 'pyze')()
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/cli/__main__.py", line 69, in main
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/cli/status.py", line 34, in run
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/cli/common.py", line 24, in get_vehicle
  File "/usr/local/lib/python3.7/dist-packages/PyZE-0.6.0-py3.7.egg/pyze/api/credentials.py", line 28, in inner
pyze.api.credentials.MissingCredentialException: kamereon-api-key

Any idea what I am doing wrong?