jamesremuscat/pyze #118: Renault switched "cockpit" to v.1

iw2been, 2024-04-27:
**Issue:**
Since last week pyze stopped working. Digging around on Internet I found that Renault switched "cockpit" back to v.1 , while pyze is using v.2.
**Fix/Workaround:**
in api/kamereon.py modify line#209in this way:
`def mileage(self):
        return self._get('cockpit', 1)`
In this way PyZE works and "pyze status" get results except:
**Other**
It seems Renault also did something about battery, 'cause pyze is not able to get percentage of charge. Modify "battery-status " to "1" or "2" doesn't solve the problem.

Badmeester, 2024-04-28:
Hi. Thanks. This indeed fixes pyze. 
In my case the percentage of charge is working just fine.
 pyze status
--------------------  ------------------------------------------------
Battery level         64%

jamesremuscat, 2024-04-29:
Already resolved in `develop` branch by 9e6a2552728d7977758cbfa7af4d6cf49c7f156a.

closed 2024-07-25 by jamesremuscat